The-OpenROAD-Project/OpenROAD #3455: OpenROAD debian installer should increment a build count or use a time stamp when naming the artifact so users know what release they have

tspyrou, 2023-06-14:
### Describe the bug

Right now the artifact has the same name.

### Expected Behavior

The name should have some indication of the date it was built on.

### Environment

```shell
git action
```

### To Reproduce

git action

oharboe, 2023-06-14:
@tspyrou How about setting up a PPA server? I believe versioning will be handled as part of this.

https://help.ubuntu.com/stable/ubuntu-help/addremove-ppa.html.en

vvbandeira, 2026-03-24:
We now provide the installer via https://vaultlink.precisioninno.com/
It contains `Version: 26Q1-951-g6975124cf2`, which has the quarter, `951` says how many commits since the tag and the git commit sha is after the `-g` -- finally, the's a link to GH with the pointer to the commit.