The-OpenROAD-Project/OpenROAD #3456: Crash in image rendering during abstract generation

oharboe, 2023-06-14:
### Describe the bug

Run:

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

Output:

```
[deleted]
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 24 u^2 17% utilization.

Signal 11 received
Stack trace:
 0# 0x00005592FCB6B8A5 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F67A863BCF0 in /lib/x86_64-linux-gnu/libc.so.6
 2# gui::DbInstDescriptor::getInstanceType(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# gui::DisplayControls::getInstRow(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# non-virtual thunk to gui::DisplayControls::isInstanceVisible(odb::dbInst*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00007F67A94CCC51 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x00007F67A8690402 in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007F67A871F590 in /lib/x86_64-linux-gnu/libc.so.6
Command terminated by signal 11
Elapsed time: 0:01.44[h:]min:sec. CPU time: user 1.30 sys 0.05 (94%). Peak memory: 259348KB.
make[1]: *** [Makefile:696: logs/asap7/mock-array-big_Element/base/6_report.log] Error 139
make[1]: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow'
make: *** [Makefile:329: results/asap7/mock-array-big_Element/base/Element.lib] Error 2
o
```

### Expected Behavior

No segfault

### Environment

```shell
$ etc/Env.sh 
Git commit: 2ccc898ac585790bc8aa1ddd9e823e2439b250cd
kernel: Linux 5.19.0-43-generic
os: Ubuntu 22.10 (Kinetic Kudu)
cmake version 3.24.2
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8554-g2ccc898ac
-- System name: Linux
-- Compiler: GNU 12.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.12.1")  
-- The C compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
[... 79 lines cut ...]
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0")  
-- boost: 1.74.0
-- Found Python3: /usr/include/python3.10 (found version "3.10.7") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Run above

vijayank88, 2023-06-14:
Is ORFS installed with `--local` or `--local --latest`?
With `--local` installation its getting stopped at placement stage.
```
Elapsed time: 0:02.17[h:]min:sec. CPU time: user 2.05 sys 0.11 (100%). Peak memory: 163348KB.
cp results/asap7/mock-array-big_Element/base/3_5_place_dp.odb results/asap7/mock-array-big_Element/base/3_place.odb
cp: \u2018results/asap7/mock-array-big_Element/base/3_5_place_dp.odb\u2019 and \u2018results/asap7/mock-array-big_Element/base/3_place.odb\u2019 are the same file
make[1]: *** [results/asap7/mock-array-big_Element/base/3_place.odb] Error 1
make[1]: Leaving directory `/home/vijayan/ORFS/OpenROAD-flow-scripts/flow'
make: *** [results/asap7/mock-array-big_Element/base/Element.lib] Error 2
```

oharboe, 2023-06-14:
> Is ORFS installed with `--local` or `--local --latest`? With `--local` installation its getting stopped at placement stage.

It doesn't crash in Docker, only with `--local` and commit: 2ccc898ac585790bc8aa1ddd9e823e2439b250cd

I'm pretty sure this has to do with the updates to drawing and that @maliberty will know what it is by looking at it. I speculate that this wasn't caught on the build server because the build server runs in Docker.

maliberty, 2023-06-14:
I ran with 
OR: 1823e2455f32c2eb5104b941e2d57bf254f44abf
ORFS: 7fd48d0e3b557156e56d4370116f1b9d465638ce

and got no crash and image files that look fine. eg:
[image]

oharboe, 2023-06-14:
@maliberty Is the error below relevant?

```
./build_openroad.sh --local --latest --openroad-args "-DCMAKE_BUID_TYPE=RELEASE -DCMAKE_CXX_FLAGS=-g"
```

ORFS: 7fd48d0e3b557156e56d4370116f1b9d465638ce
OR: 0994bfdb646b3d71842bd006f98d61c8af42908b

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk clean_all generate_abstract
[deleted]
OpenROAD v2.0-8645-g0994bfdb6
[INFO] Writing out GDS/OAS 'results/asap7/mock-array-big_Element/base/6_1_merged.gds'
Elapsed time: 0:01.02[h:]min:sec. CPU time: user 0.89 sys 0.08 (95%). Peak memory: 374432KB.
cp results/asap7/mock-array-big_Element/base/6_1_merged.gds results/asap7/mock-array-big_Element/base/6_final.gds
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/generate_abstract.tcl -metrics ./logs/asap7/mock-array-big_Element/base/generate_abstract.json) 2>&1 | tee ./logs/asap7/mock-array-big_Element/base/generate_abstract.log
OpenROAD v2.0-8645-g0994bfdb6 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Starting generation of abstract views
Elapsed time: 0:02.76[h:]min:sec. CPU time: user 2.68 sys 0.08 (100%). Peak memory: 207948KB.
``

Then:

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk clean_all generate_abstract
```

```
[deleted]
        -r ./util/def2stream.py) 2>&1 | tee ./logs/asap7/mock-array-big/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
ERROR: Unexpected end of file (line=112133, cell=MockArray, file=MockArray.lef) in Layout.read
  ./util/def2stream.py:113
Command exited with non-zero status 1
Elapsed time: 0:01.04[h:]min:sec. CPU time: user 0.94 sys 0.08 (98%). Peak memory: 377512KB.
make: *** [Makefile:750: results/asap7/mock-array-big/base/6_1_merged.gds] Error 1

```

MockArray.lef is truncated:

```
$ tail -n 10 results/asap7/mock-array-big/base/MockArray.lef 
  END io_outsUp_7[4]
  PIN io_outsUp_7[5]
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    PORT
      LAYER M5 ;
        RECT  140.16 159.756 140.184 159.84 ;
    END
  END io_outsUp_7[5]
  PIN io$
```

maliberty, 2023-06-14:
It looks like something bad happened during the generation of MockArray.lef but that seems unrelated to the image rendering.

oharboe, 2023-06-14:
Tried from top, doesn't crash.

@maliberty I'm closing, if you want to have a look at that unrelated issue, I'll leave you to it. We're not generating an abstract from mock-array-big, only mock-array-big/Element, so not urgent to fix.

oharboe, 2023-06-15:
It's back... It comes and goes...

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
[deleted]
OpenROAD v2.0-8655-g6a4ea1ea5 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO] Deleted 0 routing obstructions
[INFO RCX-0431] Defined process_corner X with ext_model_index 0
[INFO RCX-0029] Defined extraction corner X
[INFO RCX-0008] extracting parasitics of Element ...
[INFO RCX-0435] Reading extraction model file ./platforms/asap7/rcx_patterns.rules ...
[INFO RCX-0436] RC segment generation Element (max_merge_res 50.0) ...
[INFO RCX-0040] Final 818 rc segments
[INFO RCX-0439] Coupling Cap extraction Element ...
[INFO RCX-0440] Coupling threshhold is 0.1000 fF, coupling capacitance less than 0.1000 fF will be grounded.
[INFO RCX-0043] 1214 wires to be extracted
[INFO RCX-0442] 71% completion -- 865 wires have been extracted
[INFO RCX-0442] 100% completion -- 1214 wires have been extracted
[INFO RCX-0045] Extract 228 nets, 1046 rsegs, 1046 caps, 618 ccs
[INFO RCX-0015] Finished extracting Element.
[INFO RCX-0016] Writing SPEF ...
[INFO RCX-0443] 228 nets finished
[INFO RCX-0017] Finished writing SPEF ...
[INFO PSM-0083] Error file is specified as: ./reports/asap7/mock-array-big_Element/base/VDD.rpt.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0063] Specified bump pitches of 140.000 and 140.000 are less than core width of 12.096 or core height of 11.880. Changing bump location to the center of the die at (6.480, 6.480).
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[INFO PSM-0031] Number of PDN nodes on net VDD = 2268.
[INFO PSM-0064] Number of voltage sources = 1.
[INFO PSM-0040] All PDN stripes on net VDD are connected.
########## IR report #################
Worstcase voltage: 7.70e-01 V
Average IR drop  : 6.64e-05 V
Worstcase IR drop: 1.12e-04 V
######################################
[INFO PSM-0083] Error file is specified as: ./reports/asap7/mock-array-big_Element/base/VSS.rpt.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0063] Specified bump pitches of 140.000 and 140.000 are less than core width of 12.096 or core height of 11.880. Changing bump location to the center of the die at (6.480, 6.480).
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[INFO PSM-0031] Number of PDN nodes on net VSS = 2366.
[INFO PSM-0064] Number of voltage sources = 1.
[INFO PSM-0040] All PDN stripes on net VSS are connected.
########## IR report #################
Worstcase voltage: 1.59e-04 V
Average IR drop  : 7.38e-05 V
Worstcase IR drop: 1.59e-04 V
######################################

==========================================================================
finish check_setup
--------------------------------------------------------------------------
Warning: There are 7 input ports missing set_input_delay.
Warning: There are 7 output ports missing set_output_delay.
Warning: There are 7 unconstrained endpoints.

==========================================================================
finish report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
finish report_wns
[... 236 lines cut ...]
--------------------------------------------------------------------------
559.367632

==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 24 u^2 17% utilization.

Signal 11 received
Stack trace:
 0# 0x0000557D189B55C5 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FA45B63BCF0 in /lib/x86_64-linux-gnu/libc.so.6
 2# gui::DbInstDescriptor::getInstanceType(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# gui::DisplayControls::getInstRow(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# non-virtual thunk to gui::DisplayControls::isInstanceVisible(odb::dbInst*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<g [... cut ...]
 7# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00007FA45C4CCC51 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x00007FA45B690402 in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007FA45B71F590 in /lib/x86_64-linux-gnu/libc.so.6
Command terminated by signal 11
Elapsed time: 0:01.45[h:]min:sec. CPU time: user 1.25 sys 0.05 (90%). Peak memory: 259868KB.
make: *** [Makefile:696: logs/asap7/mock-array-big_Element/base/6_report.log] Error 139
```

oharboe, 2023-06-15:
@maliberty Try running `rm logs/asap7/mock-array-big_Element/base/6_report.log && make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk` several times...

QuantamHD, 2023-06-15:
@maliberty @oharboe You guys might want to try to build OpenROAD under TSAN https://clang.llvm.org/docs/ThreadSanitizer.html

oharboe, 2023-06-15:
> @maliberty @oharboe You guys might want to try to build OpenROAD under TSAN https://clang.llvm.org/docs/ThreadSanitizer.html

I tried the below, but it didn't show anything beyond the null pointer above.

```
./build_openroad.sh --local --latest --openroad-args "-D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_CXX_FLAGS=-fsanitize=address"
```

QuantamHD, 2023-06-15:
Not sure if was just a copy paste error, but the second line is also running ASAN

oharboe, 2023-06-15:
> @oharboe You might want to pass `-g -no-omit-frame-pointer` to get better line/function info in the log

Ran into escape hell :-)

```
$ ./build_openroad.sh --local --latest --openroad-args "-D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_CXX_FLAGS=\"-g -no-omit-frame-pointer -fsanitize=thread\""
[deleted]
[INFO FLW-0018] Compiling OpenROAD.
CMake Error: Unknown argument -fsanitize=thread"
CMake Error: Run 'cmake --help' for all supported options.
```

I think what @maliberty needs is to reproduce the problem on his machine and while this appear to be a race condition and not easily reproducible, hopefully the sanitizer report is easy to reproduce.

maliberty, 2023-06-15:
Something similar happened in the CI though I haven't had it happen for me.  The sanitize report doesn't even tell you which mutex or line of code is involved so its a bit frustrating.

maliberty, 2023-06-15:
I'm trying to run tsan with -g but our setup is fighting me with all sorts of compiler errors.  Would you try it if you have a chance.

maliberty, 2023-06-16:
I guessing https://github.com/The-OpenROAD-Project/OpenROAD/pull/3473 may fix this.   Please let me as I still can't reproduce the crash.  I'll be watching the CI as well.

oharboe, 2023-06-16:
```
rm logs/asap7/mock-array-big_Element/base/6_report.log && make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/final_report.tcl -metrics ./logs/asap7/mock-array-big_Element/base/6_report.json) 2>&1 | tee ./logs/asap7/mock-array-big_Element/base/6_report.log
OpenROAD v2.0-8702-gccd2fdcdc 
[deleted]
Signal 11 received
Stack trace:
 0# 0x000055CD502F72C5 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FDCA303BCF0 in /lib/x86_64-linux-gnu/libc.so.6
 2# gui::DbInstDescriptor::getInstanceType(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# gui::DisplayControls::getInstRow(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# non-virtual thunk to gui::DisplayControls::isInstanceVisible(odb::dbInst*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00007FDCA3ECCC51 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x00007FDCA3090402 in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007FDCA311F590 in /lib/x86_64-linux-gnu/libc.so.6
Command terminated by signal 11
Elapsed time: 0:01.39[h:]min:sec. CPU time: user 1.24 sys 0.06 (94%). Peak memory: 259300KB.
make: *** [Makefile:696: logs/asap7/mock-array-big_Element/base/6_report.log] Error 139
```

oharboe, 2023-06-16:
> I guessing #3473 may fix this. Please let me as I still can't reproduce the crash. I'll be watching the CI as well.

Unfortunately not it would seem. The log shows I run with a version of OpenROAD that contains the fix.

oharboe, 2023-06-16:
@maliberty Interesting... It doesn't happen when I compile in debug-mode...

```
$ ./build_openroad.sh --local --latest --openroad-args "-D CMAKE_BUILD_TYPE=DEBUG"
$ gcc --version
gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

oharboe, 2023-06-16:
Crashes:

```
./build_openroad.sh --local --latest --openroad-args "-D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_CXX_FLAGS=-O2"
```

oharboe, 2023-06-16:
Crashes:

```
./build_openroad.sh --local --latest --openroad-args "-D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_CXX_FLAGS=-O1"
```

oharboe, 2023-06-18:
Thought I saw some new changes, so tried latest:

```
rm logs/asap7/mock-array-big_Element/base/6_report.log && make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
[deleted]
OpenROAD v2.0-8762-g8d56294a5
Signal 11 received
Stack trace:
 0# 0x0000556EFED0F92B in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FB331E42520 in /lib/x86_64-linux-gnu/libc.so.6
 2# gui::DbInstDescriptor::getInstanceType(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# gui::DisplayControls::getInstRow(odb::dbInst*) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# non-virtual thunk to gui::DisplayControls::isInstanceVisible(odb::dbInst*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00007FB332CCCCA1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x00007FB331E94B43 in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007FB331F26A00 in /lib/x86_64-linux-gnu/libc.so.6
Command terminated by signal 11
Elapsed time: 0:01.14[h:]min:sec. CPU time: user 1.04 sys 0.05 (95%). Peak memory: 294168KB.
make: *** [Makefile:684: logs/asap7/mock-array-big_Element/base/6_report.log] Error 139

```

oharboe, 2023-06-18:
Using https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1143, I can build with:

```
./build_openroad.sh --local --latest --openroad-args '-D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_CXX_FLAGS="-O0 -g"'
```

@maliberty This crashes, but if I attach the debugger, it does not crash...

maliberty, 2023-06-19:
Hopefully fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/3487

oharboe, 2023-06-20:
This crashed almost every time before:

```
rm -f logs/asap7/mock-array-big_Element/base/6_report.log && make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
```

This ran without failure.

```
for i in {1..100}; do  rm -f logs/asap7/mock-array-big_Element/base/6_report.log && make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk || break;done
```

closed 2023-06-20 by oharboe