The-OpenROAD-Project/OpenROAD #3465: RePlAce divergence detected. Re-run with a smaller max_phi_cof value on small design

gadfort, 2023-06-14:
### Describe the bug

We are getting a "RePlAce divergence detected. Re-run with a smaller max_phi_cof value" error when running on a small design.
It appears all the flipflops are on top of eachother, which may be the issue.

### Expected Behavior

Completes trivial placement

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.25.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8554-g2ccc898ac
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/pgadfort/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 271 lines cut ...]
[NesterovSolve] Iter: 720 overflow: 0.441512 HPWL: 282107
[NesterovSolve] Iter: 730 overflow: 0.44109 HPWL: 283602
[NesterovSolve] Iter: 740 overflow: 0.439868 HPWL: 287017
[NesterovSolve] Iter: 750 overflow: 0.437834 HPWL: 292160
[NesterovSolve] Iter: 760 overflow: 0.434893 HPWL: 298927
[NesterovSolve] Iter: 770 overflow: 0.430677 HPWL: 307315
[NesterovSolve] Iter: 780 overflow: 0.42532 HPWL: 317399
[NesterovSolve] Iter: 790 overflow: 0.41833 HPWL: 329355
[NesterovSolve] Iter: 800 overflow: 0.409957 HPWL: 343258
[NesterovSolve] Iter: 810 overflow: 0.400042 HPWL: 359514
[NesterovSolve] Iter: 820 overflow: 0.388273 HPWL: 379323
[NesterovSolve] Iter: 830 overflow: 0.373849 HPWL: 402364
[NesterovSolve] Iter: 840 overflow: 0.357159 HPWL: 428392
[NesterovSolve] Iter: 850 overflow: 0.335507 HPWL: 457723
[NesterovSolve] Iter: 860 overflow: 0.308505 HPWL: 490127
[NesterovSolve] Iter: 870 overflow: 0.289723 HPWL: 526182
[NesterovSolve] Iter: 880 overflow: 0.271299 HPWL: 570416
[NesterovSolve] Iter: 890 overflow: 0.245248 HPWL: 621123
[NesterovSolve] Iter: 900 overflow: 0.207789 HPWL: 675015
[NesterovSolve] Iter: 910 overflow: 0.176717 HPWL: 729771
[NesterovSolve] Iter: 920 overflow: 0.161838 HPWL: 786638
[NesterovSolve] Iter: 930 overflow: 0.147445 HPWL: 833649
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: sc_place.tcl, 17 GPL-0307
```

QuantamHD, 2023-06-14:
@maliberty I thought there was a step in replace that added random noise to the instances to make sure that all the cells didn't end up on top of each other.

gadfort, 2023-06-17:
@maliberty any updates on this?

maliberty, 2023-06-19:
The clock net was getting a crazy weight which pulled all the register on top of each other (despite the small jiggle we introduce).  This is fixed in the forthcoming PR.

PR #3485 (2023-06-19): gpl: make exclusion of nets consistent in timing driven mode
Rsz excludes certain nets in findResizeSlacks (e.g. clocks) but gpl was stil querying their slack.  This lead to absurdly large net timing weights.

Fixes #3465
commit 7a3cf482 (2023-06-19): gpl: make exclusion of nets consistent in timing driven mode

Rsz excludes certain nets in findResizeSlacks (e.g. clocks) but gpl
was stil querying their slack.  This lead to absurdly large net timing
weights.

Fixes #3465
closed 2023-06-19 by maliberty