The-OpenROAD-Project/OpenROAD #3483: [ERROR DRT-0155] Guide in net net7 uses layer M6 (12) that is outside the allowed routing range [M2 (4), (M5)].

oharboe, 2023-06-19:
Following merge of https://github.com/The-OpenROAD-Project/OpenROAD/pull/3468 and using https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1097 (not merged yet), I get the error below.

Create a `settings.mk` file in the flow folder and use the updated mock-array-big https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1097:

```
export MOCK_ARRAY_PITCH_SCALE=20
export DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

Then run:

```
rm -rf results/ objects/ logs/ && make
```

Fails after 1 minute or so with output:

```
[deleted]
#macroValidPlanarAp    = 0
#macroValidViaAp       = 0
#macroNoAp             = 0
[INFO DRT-0166] Complete pin access.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:00, memory = 375.52 (MB), peak = 425.54 (MB)
[ERROR DRT-0155] Guide in net net7 uses layer M6 (12) that is outside the allowed routing range [M2 (4), (M5)].
Error: detail_route.tcl, 59 DRT-0155
Command exited with non-zero status 1
Elapsed time: 0:01.72[h:]min:sec. CPU time: user 2.81 sys 0.09 (168%). Peak memory: 435752KB.
make[1]: *** [Makefile:622: results/asap7/mock-array-big_Element/base/5_2_route.odb] Error 1
make[1]: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow'
make: *** [Makefile:329: results/asap7/mock-array-big_Element/base/Element.lib] Error 2
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/issues/3468#issuecomment-1593770094_

eder-matheus, 2023-06-19:
@oharboe In the `5_1_fastroute.log` file, I see:
```
[INFO GRT-0020] Min routing layer: M2
[INFO GRT-0021] Max routing layer: M7
```

While in `5_2_TritonRoute.log` file, we have:

`detailed_route arguments: -bottom_routing_layer M2 -top_routing_layer M5`

Looks like it's loading the file `./designs/asap7/mock-array-big/Element/config.mk` instead of `./designs/asap7/mock-array-big/config.mk`.

If I change the `MAX_ROUTING_LAYER` to M5, the detailed router ends with no errors in 6 iterations.

I'm trying to figure it out why the wrong config file is being loaded.

oharboe, 2023-06-20:
@eder-matheus https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1153 fixes the problem.

closed 2023-06-20 by maliberty