The-OpenROAD-Project/OpenROAD #3490: mpl2 crashes with mock-array-big

oharboe, 2023-06-20:
### Describe the bug

Change mock-array-big to use mpl2:

```
$ git diff
diff --git a/flow/designs/asap7/mock-array-big/config.mk b/flow/designs/asap7/mock-array-big/config.mk
index b36c175c..e0077af0 100644
--- a/flow/designs/asap7/mock-array-big/config.mk
+++ b/flow/designs/asap7/mock-array-big/config.mk
@@ -29,7 +29,8 @@ BLOCKS = Element
 
 export GDS_ALLOW_EMPTY = Element
 
-export MACRO_PLACEMENT_TCL = ./designs/asap7/mock-array-big/macro-placement.tcl
+#export MACRO_PLACEMENT_TCL = ./designs/asap7/mock-array-big/macro-placement.tcl
+export RTLMP_FLOW = True
 
 export IO_CONSTRAINTS = designs/asap7/mock-array-big/io.tcl
```

Run:

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

Output:

```
+---B2  (12)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

Determine shaping function for clusters -- Macro Tilings.

/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal /usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
6 received
Stack trace:
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x00005584892FE8FB in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FC45BE42520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00005584892FCB68 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# mpl2::SimulatedAnnealingCore<mpl2::SoftMacro>::packFloorplan() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# mpl2::SimulatedAnnealingCore<mpl2::SoftMacro>::fastSA() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00007FC45C2DC2B3 in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# 0x00007FC45BE94B43 in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007FC45BF26A00 in /lib/x86_64-linux-gnu/libc.so.6
Command terminated by signal 6
Elapsed time: 0:01.04[h:]min:sec. CPU time: user 0.91 sys 0.04 (91%). Peak memory: 193316KB.
make: *** [Makefile:476: results/asap7/mock-array-big/base/2_4_floorplan_macro.odb] Error 134
```

### Expected Behavior

No crash, hopefully macros lined up as nicely as with manual placement, since this is an array.

### Environment

```shell
latest master
```

### To Reproduce

See above

vijayank88, 2023-06-20:
`export SYNTH_HIERARCHICAL = 1`
Need to be added for RTLMP flow.

But still there is segfault.
```

maliberty, 2023-06-21:
@ZhiangWang033 I'm hoping you can help me with this.

stefanottili, 2023-07-25:
curiosity kills the cat:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/designs/asap7
doesn't contain mock-array-big, is it going to be checked in here or is it a local test case ?

oharboe, 2023-07-25:
> curiosity kills the cat: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/designs/asap7 doesn't contain mock-array-big, is it going to be checked in here or is it a local test case ?

mock-array-big was renamed to mock-array.

stefanottili, 2023-08-15:
I've been running mock-array on a M1 MacBook Ventura and besides https://github.com/The-OpenROAD-Project/OpenROAD/issues/3847 it run's thru.

This can be closed, no ?

closed 2023-08-15 by maliberty