The-OpenROAD-Project/OpenROAD #3492: Unable to use bond_pad

vijayank88, 2023-06-20:
### Describe the bug

I am trying to implement `place_pad` command with simple gcd design. 
But while implementing `bond_pad` it throws following error:
```
number instances in verilog is 288
[WARNING IFP-0028] Core area lower left (210.000, 210.000) snapped to (210.220, 212.160).
[INFO IFP-0001] Added 414 rows of 2456 site unithd with height 1.
[ERROR PAD-0011] sky130_ef_io__bare_pad is not of type COVER, but is instead BLOCK
[ERROR GUI-0070] Error: pad.tcl, 74 PAD-0011
```

### Expected Behavior

Complete pad level connections

### Environment

```shell
Latest master commit
```

### To Reproduce

[floorplan_gcd_sky130hd_base_2023-06-20_13-18.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11800914/floorplan_gcd_sky130hd_base_2023-06-20_13-18.tar.gz)

Extract and run `./run.sh`

vijayank88, 2023-06-20:
Without `bond_pad` it fails at GPL stage.
```
[NesterovSolve] Iter: 340 overflow: 0.252138 HPWL: 29827920
[NesterovSolve] Iter: 350 overflow: 0.237423 HPWL: 29295647
[NesterovSolve] Iter: 360 overflow: 0.208085 HPWL: 29375424
[INFO GPL-0100] worst slack 5.68e-10
[INFO GPL-0103] Weighted 29 nets.
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 51 GRT-0042
Elapsed time: 0:15.94[h:]min:sec. CPU time: user 15.82 sys 0.11 (99%). Peak memory: 145368KB.
make: *** [Makefile:529: results/sky130hd/gcd/base/3_3_place_gp.odb] Error 1

```

vijayank88, 2023-06-20:
@gadfort 
After changing to `COVER` it fails at pdn stage for gcd design.
```
[WARNING PDN-0189] Supply pin VCCHIB of instance IO_FILL_IO_SOUTH_0_9 is not connected to any net.
[WARNING PDN-0189] Supply pin VDDIO of instance IO_FILL_IO_SOUTH_0_9 is not connected to any net.
[WARNING PDN-0189] message limit reached, this message will no longer print
[INFO PDN-0001] Inserting grid: grid
[WARNING PDN-0178] Remaining channel (535.1000, 211.9200) - (631.6900, 215.3650) on met4 for nets: VSS, VDD
[WARNING PDN-0178] Remaining channel (779.3600, 211.9200) - (875.9500, 215.3650) on met4 for nets: VSS, VDD
[ERROR PDN-0179] Unable to repair all channels.
Error: pdn.tcl, 5 PDN-0179
Elapsed time: 0:01.09[h:]min:sec. CPU time: user 0.99 sys 0.09 (99%). Peak memory: 109768KB.
make: *** [Makefile:487: results/sky130hd/gcd/base/2_6_floorplan_pdn.odb] Error 1
```

vijayank88, 2023-06-20:
Missed to add `connect_by_abutment` in `pad.tcl`. Now 
```
omponents of this program may be licensed under more restrictive licenses which must be honored.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_vzz_0 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_vzz_0 is not connected to any net.
[WARNING PDN-0189] Supply pin DRN_HVC of instance u_v18_0 is not connected to any net.
[WARNING PDN-0189] Supply pin SRC_BDY_HVC of instance u_v18_0 is not connected to any net.
[INFO PDN-0001] Inserting grid: grid
[WARNING PDN-0178] Remaining channel (535.1000, 211.9200) - (631.6900, 215.3650) on met4 for nets: VSS, VDD
[WARNING PDN-0178] Remaining channel (779.3600, 211.9200) - (875.9500, 215.3650) on met4 for nets: VSS, VDD
[ERROR PDN-0179] Unable to repair all channels.
Error: pdn.tcl, 5 PDN-0179
Elapsed time: 0:01.08[h:]min:sec. CPU time: user 1.00 sys 0.08 (99%). Peak memory: 109768KB.
make: *** [Makefile:487: results/sky130hd/gcd/base/2_6_floorplan_pdn.odb] Error 1
```

gadfort, 2023-06-20:
The bondpads are not aligned properly, so they stick into the design area.

[image]

vijayank88, 2023-06-21:
@gadfort 
I have remove `-offset`, now it is passed pdn stage. But fail at global placement.
```
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 51 GRT-0042
Elapsed time: 0:16.40[h:]min:sec. CPU time: user 16.28 sys 0.11 (99%). Peak memory: 154780KB.
make: *** [Makefile:529: results/sky130hd/gcd/base/3_3_place_gp.odb] Error 1
```
[global_place_gcd_sky130hd_base_2023-06-21_05-34.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11813827/global_place_gcd_sky130hd_base_2023-06-21_05-34.tar.gz)

maliberty, 2023-06-21:
Have you looked at this pin?  The message could be better.

vijayank88, 2023-06-22:
[image]
Here is the close view of clk pin.

maliberty, 2023-06-22:
I don't know where the pin is in that image.  What layer(s) are the pin shapes on?

vijayank88, 2023-06-22:
If you run latest test case able to view in gui. It's highlights in all layer if I disable layer by layer.

maliberty, 2023-06-22:
It looks to me like clk has no shapes:
[image]

maliberty, 2023-06-22:
```
PINS 54 ;
    - clk + NET clk + DIRECTION INPUT + USE SIGNAL ;
```
confirms it

vijayank88, 2023-06-22:
> ```
> PINS 54 ;
>     - clk + NET clk + DIRECTION INPUT + USE SIGNAL ;
> ```
> 
> confirms it

All other `PINS` too same format:
```
PINS 54 ;
    - clk + NET clk + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[0] + NET req_msg[0] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[10] + NET req_msg[10] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[11] + NET req_msg[11] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[12] + NET req_msg[12] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[13] + NET req_msg[13] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[14] + NET req_msg[14] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[15] + NET req_msg[15] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[16] + NET req_msg[16] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[17] + NET req_msg[17] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[18] + NET req_msg[18] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[19] + NET req_msg[19] + DIRECTION INPUT + USE SIGNAL ;
    - req_msg[1] + NET req_msg[1] + DIRECTION INPUT + USE SIGNAL ;
```
Where it should get updated?

maliberty, 2023-06-22:
I don't really know this design.  It seems these signals should be going through the pads.

vijayank88, 2023-06-23:
> I don't really know this design.  It seems these signals should be going through the pads.

I meant asking something we need to update our flow script to support chip level connections?

maliberty, 2023-06-23:
I suspect it is a configuration or netlist issue rather than a flow one.

vijayank88, 2023-06-28:
Test case:
```
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 51 GRT-0042
Elapsed time: 3:39.42[h:]min:sec. CPU time: user 218.83 sys 0.55 (99%). Peak memory: 704336KB.
make: *** [Makefile:538: results/sky130hd/gcd/base/3_3_place_gp.odb] Error 1
```
[global_place_gcd_sky130hd_base_2023-06-28_11-15.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11893509/global_place_gcd_sky130hd_base_2023-06-28_11-15.tar.gz)

vijayank88, 2023-07-10:
@maliberty 
`FOOTPRINT_TCL` called in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/4315c6cea124a86a1bcbe69c08dcbf6a2c587514/flow/scripts/floorplan.tcl#L64

In `3_3_place_gp.log` getting following warnings
```
OpenROAD v2.0-9070-gbb341ffb9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 210220 212160
[INFO GPL-0005] CoreAreaUxUy: 1669800 1667360
[WARNING GPL-0001] clk toplevel port is not placed!
       Replace will regard clk is placed in (0, 0)
[WARNING GPL-0001] req_msg[0] toplevel port is not placed!
       Replace will regard req_msg[0] is placed in (0, 0)
[WARNING GPL-0001] req_msg[10] toplevel port is not placed!
       Replace will regard req_msg[10] is placed in (0, 0)
[WARNING GPL-0001] req_msg[11] toplevel port is not placed!
       Replace will regard req_msg[11] is placed in (0, 0)
[WARNING GPL-0001] req_msg[12] toplevel port is not placed!
       Replace will regard req_msg[12] is placed in (0, 0)
[WARNING GPL-0001] req_msg[13] toplevel port is not placed!
       Replace will regard req_msg[13] is placed in (0, 0)
[WARNING GPL-0001] req_msg[14] toplevel port is not placed!
       Replace will regard req_msg[14] is placed in (0, 0)
[WARNING GPL-0001] req_msg[15] toplevel port is not placed!
       Replace will regard req_msg[15] is placed in (0, 0)
[WARNING GPL-0001] req_msg[16] toplevel port is not placed!
       Replace will regard req_msg[16] is placed in (0, 0)
[WARNING GPL-0001] req_msg[17] toplevel port is not placed!
       Replace will regard req_msg[17] is placed in (0, 0)
[WARNING GPL-0001] req_msg[18] toplevel port is not placed!
       Replace will regard req_msg[18] is placed in (0, 0)
[WARNING GPL-0001] req_msg[19] toplevel port is not placed!
       Replace will regard req_msg[19] is placed in (0, 0)
[WARNING GPL-0001] req_msg[1] toplevel port is not placed!
       Replace will regard req_msg[1] is placed in (0, 0)
[WARNING GPL-0001] req_msg[20] toplevel port is not placed!
       Replace will regard req_msg[20] is placed in (0, 0)
[WARNING GPL-0001] req_msg[21] toplevel port is not placed!
       Replace will regard req_msg[21] is placed in (0, 0)
[WARNING GPL-0001] req_msg[22] toplevel port is not placed!
       Replace will regard req_msg[22] is placed in (0, 0)
[WARNING GPL-0001] req_msg[23] toplevel port is not placed!
       Replace will regard req_msg[23] is placed in (0, 0)
[WARNING GPL-0001] req_msg[24] toplevel port is not placed!
       Replace will regard req_msg[24] is placed in (0, 0)
[WARNING GPL-0001] req_msg[25] toplevel port is not placed!
       Replace will regard req_msg[25] is placed in (0, 0)
[WARNING GPL-0001] req_msg[26] toplevel port is not placed!
       Replace will regard req_msg[26] is placed in (0, 0)
[WARNING GPL-0001] req_msg[27] toplevel port is not placed!
       Replace will regard req_msg[27] is placed in (0, 0)
[WARNING GPL-0001] req_msg[28] toplevel port is not placed!
       Replace will regard req_msg[28] is placed in (0, 0)
[WARNING GPL-0001] req_msg[29] toplevel port is not placed!
       Replace will regard req_msg[29] is placed in (0, 0)
[WARNING GPL-0001] req_msg[2] toplevel port is not placed!
       Replace will regard req_msg[2] is placed in (0, 0)
[WARNING GPL-0001] req_msg[30] toplevel port is not placed!
       Replace will regard req_msg[30] is placed in (0, 0)
[WARNING GPL-0001] req_msg[31] toplevel port is not placed!
       Replace will regard req_msg[31] is placed in (0, 0)
[WARNING GPL-0001] req_msg[3] toplevel port is not placed!
       Replace will regard req_msg[3] is placed in (0, 0)
[WARNING GPL-0001] req_msg[4] toplevel port is not placed!
       Replace will regard req_msg[4] is placed in (0, 0)
[WARNING GPL-0001] req_msg[5] toplevel port is not placed!
       Replace will regard req_msg[5] is placed in (0, 0)
[WARNING GPL-0001] req_msg[6] toplevel port is not placed!
       Replace will regard req_msg[6] is placed in (0, 0)
[WARNING GPL-0001] req_msg[7] toplevel port is not placed!
       Replace will regard req_msg[7] is placed in (0, 0)
[WARNING GPL-0001] req_msg[8] toplevel port is not placed!
       Replace will regard req_msg[8] is placed in (0, 0)
[WARNING GPL-0001] req_msg[9] toplevel port is not placed!
       Replace will regard req_msg[9] is placed in (0, 0)
[WARNING GPL-0001] req_rdy toplevel port is not placed!
       Replace will regard req_rdy is placed in (0, 0)
```
That means IO's are not placed. Then it fails
```
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 51 GRT-0042
Elapsed time: 3:47.54[h:]min:sec. CPU time: user 226.55 sys 0.60 (99%). Peak memory: 704204KB.
```
We need to call pin placement along with `FOOTPRINT_TCL` or move after https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/io_placement_random.tcl will resolve `[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.`

gadfort, 2023-07-10:
I don't think you will want to run pin-placement as that will not correctly handle the port IOs. Something looks to have gone wrong when placing the bond pads which should have created the bpins and marked these signals as special. Is it possible that the pads are not actually correctly connected to the circuit?

vijayank88, 2023-07-11:
@gadfort 
I don't know where I'm doing mistake. Created `pad.tcl` for gcd. Source files here: 
[gcd_pad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12013358/gcd_pad.zip)

Steps to run the flow:
```
1. unzip gcd_pad.zip `
2. cd OpenROAD-flow-scripts/flow
3. move `gcd_pad` inside `designs/sky130hd/
4. make DESIGN_CONFIG=./designs/sky130hd/gcd_pad/config.mk
```
It will reproduce the error.
Fixing this issue will be a good example for IO pad based connections.

vijayank88, 2023-07-19:
@gadfort 
Have you had chance to look into this design?

vijayank88, 2023-09-07:
@gadfort 
Can you help me to complete this test case?

vijayank88, 2024-03-07:
Able to pass coyote_tc design for IO pad placement. coyote_tc is the main focus. So closing this issue.

closed 2024-03-07 by vijayank88