The-OpenROAD-Project/OpenROAD #3494: grt heap underflow in aes_core OL CI

maliberty, 2023-06-21:
### Describe the bug

[ERROR GRT-0183] Net net767: heap underflow during 3D maze routing.

from https://jenkins.openroad.tools/job/OpenLane-Public/2142/execution/node/469/log/

### Expected Behavior

No error in grt

### Environment

```shell
See jenkins
```

### To Reproduce

See jenkins CI

maliberty, 2023-06-21:
@eder-matheus feel free to reassign if needed

vijayank88, 2023-06-21:
I ran `aes_core` with latest commit of OpenLane locally. Not failed during GRT. Maybe it is specific to CI.

maliberty, 2023-06-21:
I believe the CI uses the head of OR master.  I'm guessing that OL is behind that.

kareefardi, 2023-07-04:
Any update on this ?