The-OpenROAD-Project/OpenROAD #3502: CTS: Attempt to create instance with duplicate name

kareefardi, 2023-06-22:
### Describe the bug

Error `[ERROR ODB-0385] Attempt to create instance with duplicate name: clkbuf_0_core_clk` during CTS. 

### Expected Behavior

Might be an input fault, but it is not clear how to proceed. 

### OpenROAD Environment

```shell
Git commit: 7586d03811ca43cc2615077afb0f4821c81f82e2
kernel: Linux 6.1.0-1014-oem
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8804-g7586d0381
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/karim/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
[... 162 lines cut ...]
[INFO CTS-0018]     Created 3 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 3 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 2:1, 5:1..
[INFO CTS-0017]     Max level of the clock tree: 1.
[INFO CTS-0018]     Created 3 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 3 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 8:1, 12:1..
[INFO CTS-0017]     Max level of the clock tree: 1.
[INFO CTS-0018]     Created 3 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 3 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 3:1, 4:1..
[INFO CTS-0017]     Max level of the clock tree: 1.
[ERROR ODB-0385] Attempt to create instance with duplicate name: clkbuf_0_core_clk
Error: cts.tcl, 59 ODB-0385
```

### Additional Context

OpenLane is not using the latest OpenROAD. However, when ran outside OpenLane, using the latest OpenROAD, same error appeared.

maliberty, 2023-06-22:
My guess is the generated clock
```
create_generated_clock -name core_clk -source [get_pins {_208_/X}] -divide_by 1 [get_ports {core_clk}]
```
but I'll take a look.

maliberty, 2023-06-22:
I don't understand what you are trying to do with your generated clock statement.  It seems odd...

PR #3509 (2023-06-23): cts: avoid revisiting a generated clock
Fixes #3502
commit 146871de (2023-06-23): cts: avoid revisiting a generated clock

Fixes #3502
closed 2023-06-23 by maliberty