The-OpenROAD-Project/OpenROAD #3511: Remove DRC errors in small gaps by using straight wires

oharboe, 2023-06-23:
### Description

Use https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1174, then create a `flow/settings.mk` file:

```
export DESIGN_CONFIG=designs/asap7/mock-array/config.mk
# Routing by abutment parameters
export MOCK_ARRAY_TABLE         = 8 8 4 4 4.11111111111111 4.11111111111111
export MACRO_BLOCKAGE_HALO      = 0
```

Run make and look at the DRC errors:

[image]

Zooming in. If this was just routed as a straight wire, I don't know if that's possible, then my hope is that the DRC errors would go away.

[image]

### Suggested Solution

Naively: use straight wires in tiny gaps in detailed routing.

maliberty, 2023-06-24:
It would be better if the pins were on the routing tracks:
[image]

oharboe, 2023-06-24:
> It would be better if the pins were on the routing tracks:

What is the positioning constraint?

I am using manual placement and I have tried to place the macro "on the right grid'?

maliberty, 2023-06-24:
The centerline of the pin should fall on the track ideally so that that are wire on track will go directly into the pin.  Generally the router tries to use the tracks.

oharboe, 2023-06-24:
> The centerline of the pin should fall on the track ideally so that that are wire on track will go directly into the pin. Generally the router tries to use the tracks.

I've been using multiples of 2.16 (45 * 0.048), isn't that correct?

What else should I use?

maliberty, 2023-06-26:
I'm guessing that 4.11111111111111 doesn't align to the tracks.  Where did that value come from?

maliberty, 2023-06-26:
13.359 is not a multiple of 2.16:
[image]

maliberty, 2023-06-26:
I'm not sure but in your diff you changed int to float.  You never want to use floats for location calculations due to rounding issues.  Integer DBUs are a best practice.

maliberty, 2023-06-26:
I think this is more of an issue in the construction of the array than in the router.

oharboe, 2023-06-27:
> I think this is more of an issue in the construction of the array than in the router.

Agreed.

I will rework mock array to use snaller gaps and integers and check that pins are centered.