The-OpenROAD-Project/OpenROAD #3518: openroad installation Lemon package

msingh9, 2023-06-24:
### Describe the bug

It seems like lemon source can't be downloaded.

--2023-06-23 20:36:52--  (try: 2)  http://lemon.cs.elte.hu/pub/sources/lemon-1.3.1.tar.gz
Connecting to lemon.cs.elte.hu (lemon.cs.elte.hu)|157.181.227.195|:80... failed: Connection timed out.
Retrying.

I have tried, http://lemon.cs.elte.hu is not reachable. 

Is there a place I can get this from? I was successful in installing it last time (few months ago), but don't remember where did I get lemon from.

I do have one lemon installed but that is the wrong one.

### Expected Behavior

lemon installation fails.

### Environment

```shell
Ubuntu 20.04 running under Microsoft WSL2
```

### To Reproduce

wget http://lemon.cs.elte.hu/pub/sources/lemon-1.3.1.tar.gz

maliberty, 2023-06-24:
This was recently resolved by putting a copy in GH.  Please update and try again.

vijayank88, 2023-06-24:
@msingh9  

Try below commands: 
```
git checkout master
git pull
./etc/Build.sh --clean
```

msingh9, 2023-06-25:
Thank you @vijayank88 and @maliberty . I am past that issue. I saw this warning. while installation. I am not sure if this will be problem, if so, how do I fix this.

CMake Warning at src/CMakeLists.txt:276 (add_executable):
  Cannot generate a safe runtime search path for target openroad because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libtcl8.6.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /root/miniconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /root/miniconda3/lib
    runtime library [libgomp.so.1] in /usr/lib/gcc/x86_64-linux-gnu/11 may be hidden by files in:
      /root/miniconda3/lib

  Some of these libraries may not be found correctly.

vvbandeira, 2023-06-25:
@msingh9, could you try again but disable any python environment before trying to build?

---

@luarss, I think we should add a warning about not using python environments during the build process.

msingh9, 2023-06-25:
Okay, It failed so I am trying again with disabling python environment.

[ 78%] Linking CXX static library libabc.a
gmake[1]: *** [CMakeFiles/Makefile2:5017: src/gui/CMakeFiles/gui.dir/all] Error 2
[ 78%] Built target libabc
gmake: *** [Makefile:146: all] Error 2

msingh9, 2023-06-26:
I don't see that warning once python env is disabled. But now I hit another error.

/mnt/c/Users/msingh/OpenROAD/src/gui/src/staGui.cpp: In member function ‘virtual void gui::TimingConeRenderer::drawObjects(gui::Painter&)’:
/mnt/c/Users/msingh/OpenROAD/src/gui/src/staGui.cpp:825:30: error: ‘class sta::Unit’ has no member named ‘scaleAbbreviation’; did you mean ‘scaleAbreviation’?
  825 |         = std::string(units->scaleAbbreviation()) + units->suffix();
      |                              ^~~~~~~~~~~~~~~~~
      |                              scaleAbreviation
[ 77%] Building CXX object src/gui/CMakeFiles/gui.dir/src/ruler.cpp.o
gmake[2]: *** [src/gui/CMakeFiles/gui.dir/build.make:386: src/gui/CMakeFiles/gui.dir/src/staGui.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:5017: src/gui/CMakeFiles/gui.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

maliberty, 2023-06-26:
Most likely your sta submodule is out of date, try: git submodule update --init --recursive

msingh9, 2023-06-26:
I am hitting another error now.

msingh9, 2023-06-26:
/mnt/c/Users/msingh/OpenROAD/src/odb/test/cpp/TestAbstractLef.cc:13:10: fatal error: gmock/gmock.h: No such file or directory
   13 | #include "gmock/gmock.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/odb/test/cpp/CMakeFiles/OdbGTests.dir/build.make:90: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/TestAbstractLef.cc.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[ 71%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/ivy/ivyRwr.c.o
[ 71%] Building CXX object src/rcx/src/CMakeFiles/rcx_lib.dir/extCoords.cpp.o
[ 71%] Building CXX object src/rcx/src/CMakeFiles/rcx_lib.dir/extFlow.cpp.o
[ 71%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/ivy/ivySeq.c.o
[ 71%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/ivy/ivyShow.c.o
gmake[1]: *** [CMakeFiles/Makefile2:2303: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/all] Error 2

vijayank88, 2023-06-26:
Can you point the link which you're using for the installation?
 @msingh9

vvbandeira, 2023-06-26:
> 13 | #include "gmock/gmock.h"
> | ^~~~~~~~~~~~~~~

@msingh9, please update to the latest version. This error has been addressed in #3514

msingh9, 2023-06-26:
I am using https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/docs/user/Build.md.
./etc/Build.sh -clean 
I pulled everything from master branch at  https://github.com/The-OpenROAD-Project/OpenROAD.git.
Trying again..

msingh9, 2023-06-26:
Just an update, not sure if this is the problem at my end or anyone having the same issue. Now it is getting stuck at around 67% of installation. I am running under windows WSL. 

[ 67%] Building CXX object src/rcx/src/CMakeFiles/rcxUnitTest.dir/__/test/ext2dBoxTest.cpp.o
[ 67%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/saig/saigSynch.c.o
[ 67%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/saig/saigTempor.c.o
[ 67%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/saig/saigTrans.c.o
[ 67%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/aig/saig/saigWnd.c.o
...

vvbandeira, 2023-06-26:
@msingh9 could you upload the `build/openroad_build.log` file? Also, depending on your HW it might be a good idea to try with `-threads=1`

msingh9, 2023-06-27:
It worked finally. Issue was on my machine in mounting some network drive which will time out and cause the hang. After disabling that mount, I was successfully able to build it. Only strange thing was the Openroad executable was not automatically copied to install directory, rather I had to manually copy from build/src/openroad. I was building this under WSL2 on windows.

closed 2023-06-27 by msingh9

vvbandeira, 2023-06-27:
@msingh9, that's great news. If you followed any guide/tutorial on the web, it could be beneficial to post it here so that if someone else gets a similar problem, we have documented it.

msingh9, 2023-06-28:
This is how I installed on my Windows-10 pro machine.
1) Install WSL2 and Ubuntu 22.04 [Don't remember the detailed steps, but I followed this https://learn.microsoft.com/en-us/windows/wsl/install]
2. I got hit by this issue (https://github.com/MicrosoftDocs/WSL/issues/547) which I resolved by issuing command 3/4 below.
3. C:\Windows\system32> wsl --shutdown
4. C:\Windows\system32> netsh winsock reset
5. Now Ubuntu runs. You may have to reset and run those command with admin privilege. 
6. I had external network drive mounted in ubuntu, which make my installation hang. So, make sure unmount any network drive. you can mount them later.
7. Then just follow installation steps from https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/docs/user/Build.md.
8. I did 
 git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
 cd OpenROAD
 mkdir build
 cd build
 cmake ..
 make
11. openroad executable is in build/src/ directory. I copied to /usr/local/bin manually.
12. 
This is it.

maliberty, 2023-06-28:
"make install" will copy it