The-OpenROAD-Project/OpenROAD #3520: Unable to build due to missing gmock/gmock.h

gadfort, 2023-06-24:
### Describe the bug

Since https://github.com/The-OpenROAD-Project/OpenROAD/pull/3508 I'm unable to build normally due a missing header file that does not appear to get installed by the DependencyInstaller nor is automatically captured by the cmake googletest install.

'''
[ 25%] Built target rcx_lib
[ 27%] Building CXX object src/dpl/test/CMakeFiles/dpl_test.dir/dpl_test.cc.o
[ 27%] Swig compile ../common/odb.i for python
[ 27%] Building CXX object src/odb/test/cpp/CMakeFiles/TestMaster.dir/TestMaster.cpp.o
/home/pgadfort/OpenROAD/src/odb/test/cpp/TestAbstractLef.cc:13:10: fatal error: gmock/gmock.h: No such file or directory
   13 | #include "gmock/gmock.h"
      |          ^~~~~~~~~~~~~~~
'''

If I turn off building with tests I'm able to 

### Expected Behavior

OpenROAD builds correctly

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.25.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8836-g0ea657420
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/pgadfort/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 83 lines cut ...]
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.0")
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.e0UGkVHmmO
```

### To Reproduce

cmake ..
make

gadfort, 2023-06-24:
@QuantamHD FYI, I'm not sure why this is happening since it looks like googletest is present

QuantamHD, 2023-06-24:
-- Found GTest: /home/pgadfort/.local/lib/cmake/GTest/GTestConfig.cmake

That looks like it might be picking up the system gtest.

Can you try to remove this line https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/third-party/gtest/CMakeLists.txt#L1 

Then remove your build folder and try again?

gadfort, 2023-06-24:
@QuantamHD that seems to work for me. Is this related to: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3514?
It seems like the solution would be to not use the system gtest if that can cause problems.

QuantamHD, 2023-06-24:
Yeah I think we need to remove it. It seems like a lot of the OS packages are incomplete or broken.

closed 2023-06-25 by vvbandeira