The-OpenROAD-Project/OpenROAD #3529: pad/test unit test fails

vijayank88, 2023-06-26:
### Describe the bug

`skywater130_coyote_tc.tcl` unit test fails
```
[ERROR PAD-0001] Unable to place u_rocc_mem_req_data_i.u_io\[13\].u_in (sky130_fd_io__top_gpiov2) at (5121.000um, 319.665um) - (5201.000um, 519.665um) as it overlaps with u_rocc_mem_req_data_i.u_io\[12\].u_in (sky130_fd_io__top_gpiov2)
Error: skywater130_coyote_tc.tcl, 109 PAD-0001
while evaluating {source skywater130_coyote_tc.tcl }
```
`skywater130_caravel.tcl` unit test fails
```
[ERROR PAD-0001] Unable to place brk_vccd_0 (sky130_ef_io__disconnect_vccd_slice_5um) at (3384.000um, 355.000um) - (3389.000um, 552.965um) as it overlaps with brk_vdda_0 (sky130_ef_io__disconnect_vdda_slice_5um)
Error: skywater130_caravel.tcl, 131 PAD-0001
```

### Expected Behavior

Pass unit test

### Environment

```shell
Git commit: 8aabb2ef822bacd5a6710da4b9c269e72f345ea3
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
CMake Error at CMakeLists.txt:112 (message):
  Insufficient gcc version.  Found 4.8.5, but require >= 8.3.0.

-- The CXX compiler identification is GNU 4.8.5
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8870-g8aabb2ef8
-- Configuring incomplete, errors occurred!
```

### To Reproduce

```
cd src/pad/test
openroad
source skywater130_caravel.tcl
```

```
source skywater130_coyote_tc.tcl
```

### Relevant log output

```shell
Refer `describe bug` section
```

maliberty, 2023-06-26:
I see no error with 8aabb2ef822bacd5a6710da4b9c269e72f345ea3 (head).

Also
```
CMake Error at CMakeLists.txt:112 (message):
  Insufficient gcc version.  Found 4.8.5, but require >= 8.3.0.
```
looks concerning.  4.8.5 is ancient.

vijayank88, 2023-06-26:
Recently this GCP machine got restarted. I need to use 
```
# to enable gcc-8
source /opt/rh/devtoolset-8/enable
# or to enable clang-7
source /opt/rh/llvm-toolset-7.0/enable
```
Now it is working fine.

closed 2023-06-26 by vijayank88

maliberty, 2023-06-26:
I put those in my .bashrc.