The-OpenROAD-Project/OpenROAD #3532: Different messages with the same ID

vvbandeira, 2023-06-26:
### Describe the bug

DFT is missing the `messages(dtf)` call in CMake and has duplicated IDs.
MPL is repeating messages across `mpl` and `mpl2`; should this be allowed?
RSZ and dbSta have ORD messages, is this expected? Also, there are duplicate IDs.

### Expected Behavior

Do not repeat message IDs.

### Environment

```shell
Latest commit.
```

### To Reproduce

```
cd src && ../etc/find_messages.py
```

Any reason for not running this instead of running it on each folder separately?

### Relevant log output

```shell
> cd src && ../etc/find_messages.py > messages.txt
Error: DFT 0004 used 2 times
Error: DFT 0005 used 2 times
Error: MPL 0002 used 2 times
Error: MPL 0003 used 2 times
Error: MPL 0004 used 2 times
Error: MPL 0005 used 2 times
Error: MPL 0009 used 2 times
Error: ORD 0101 used 2 times
Error: ORD 0102 used 2 times
Error: ORD 0105 used 2 times
Error: ORD 1009 used 2 times
Error: ORD 1010 used 2 times
Error: ORD 1011 used 2 times
Error: ORD 1012 used 2 times
Error: ORD 1013 used 2 times

DFT 0004 ClockDomain.cpp:52        Clock mix config requested is not supported
DFT 0004 ScanCellFactory.cpp:137   Cell '{:s}' doesn't have a valid clock connected. Can't create a scan cell
DFT 0005 ScanCellFactory.cpp:146   Cell '{:s}' is not a scan cell. Can't use it for scan architect
DFT 0005 dft.i:80                  Requested clock mixing config not valid

ORD 0101 Design.cc:80              Only one of the options -incremental and -floorplan_init can be set at a time
ORD 0101 Resizer.tcl:48            Use -layer or -via but not both.
ORD 0102 Design.cc:86              No technology has been read.
ORD 0102 Resizer.tcl:57            layer $layer_name not found.
ORD 0105 OpenRoad.cc:473           Can't open {}
ORD 0105 Resizer.tcl:98            via $layer_name not found.
ORD 1009 OpenRoad.tcl:193          -name is missing.
ORD 1009 dbReadVerilog.tcl:51      missing top_cell_name argument and no current_design.
ORD 1010 OpenRoad.tcl:196          Either -net or -all_clocks need to be defined.
ORD 1010 dbReadVerilog.tcl:58      no technology has been read.
ORD 1011 OpenRoad.tcl:202          No NDR named ${ndrName} found.
ORD 1011 dbReadVerilog.cc:425      LEF master {} has no liberty cell.
ORD 1012 OpenRoad.tcl:208          No net named ${netName} found.
ORD 1012 dbReadVerilog.cc:431      Liberty cell {} has no LEF master.
ORD 1013 OpenRoad.tcl:155          -masters is required.
ORD 1013 dbReadVerilog.cc:293      instance {} LEF master {} not found.

MPL 0002 MacroPlacer.cpp:164       Some instances do not have Liberty models. TritonMP will place macros without connection information.
MPL 0002 hier_rtlmp.cpp:867        Floorplan has not been initialized? Pin location error for {}.
MPL 0003 MacroPlacer.cpp:808       Macro {} is unplaced, use global_placement to get an initial placement before macro placement.
MPL 0003 hier_rtlmp.cpp:2577       There are no valid tilings for mixed cluster: {}
MPL 0004 MacroPlacer.cpp:829       No macros found.
MPL 0004 hier_rtlmp.cpp:2800       This no valid tilings for hard macro cluser: {}
MPL 0005 MacroPlacer.cpp:833       Found {} macros.
MPL 0005 hier_rtlmp.cpp:3549       [MultiLevelMacroPlacement] Failed on cluster: {}
MPL 0009 MacroPlacer.cpp:196       {} pins {}.
MPL 0009 hier_rtlmp.cpp:4989       Fail !!! Bus planning error !!!
```

maliberty, 2023-06-26:
@fgaray please fix the DFT messages issue.

I think we can leave mpl/mpl2 as we plan to remove mpl.  This is further motivation to finish it off.

rsz's ORD are in set_layer_rc which points to the fact that it doesn't really belong in rsz.  Its not rsz specific info and grt also relies on it.

dbSta is always a bit odd as its not really its own module nor a part of sta.  Maybe its time to give it its own code.   Until then ORD is as good as anything.  Using STA might conflict with sta/

vvbandeira, 2023-06-26:
> I think we can leave mpl/mpl2 as we plan to remove mpl. This is further motivation to finish it off.

Sounds good.

> rsz's ORD are in set_layer_rc which points to the fact that it doesn't really belong in rsz. Its not rsz specific info and grt also relies on it.

Should we move this to `utl` maybe?

> dbSta is always a bit odd as its not really its own module nor a part of sta. Maybe its time to give it its own code. Until then ORD is as good as anything. Using STA might conflict with sta/

Should we just update the IDs at least?

Let me know if I should start any PRs. Also, should we be running the following at least on CI to capture similar issues?
```
cd src && ../etc/find_messages.py
```

maliberty, 2023-06-26:
>Should we move this to utl maybe?

I think odb would be better as it avoids the need for loading setRC.tcl in various places through the flow.
> Should we just update the IDs at least?

Yes, please make a PR for this.

> cd src && ../etc/find_messages.py

Seems like a good check in the CI

closed 2023-06-26 by vvbandeira