The-OpenROAD-Project/OpenROAD #3556: by default openroad is using 2+16 threads for detail_route on a 8 core M1

stefanottili, 2023-06-29:
### Describe the bug

I noticed longer runtimes using openroad vs openroad -threads 8 on a 8 core (4+4) M1.
53min, 52min vs 40min, 45min

The Activity Monitor shows openroad using 18 threads vs 10

a) openroad by default uses 18 threads, 2 before drt, 2 + 16 = 18 during drt
b) openroad -threads 8 uses 10 threads, 2 before drt, 2 + 8 = 10 during drt

OpenRoad is reporting to be using 8 threads in both cases.
```
openroad (Activity Monitor shows 2+16)

  1 OpenROAD v2.0-8931-g74ada9e7c 
  4 [INFO ODB-0222] Reading LEF file: ispd19_test6/ispd19_test6.input.lef.gz
 17 [INFO ODB-0134] Finished DEF file: ispd19_test6/ispd19_test6.input.def.gz            
 18 [INFO ORD-0030] Using 8 thread(s).
```
```
openroad -threads 8 (Activity Monitor shows 2+8)

  1 OpenROAD v2.0-8931-g74ada9e7c
  4 [INFO ORD-0030] Using 8 thread(s).
  5 [INFO ODB-0222] Reading LEF file: ispd19_test6/ispd19_test6.input.lef.gz
 18 [INFO ODB-0134] Finished DEF file: ispd19_test6/ispd19_test6.input.def.gz
 19 [INFO ORD-0030] Using 8 thread(s).
```

### Expected Behavior

use 8 threads for drt by default on a 8 core cpu, not 16

### Environment

```shell
Jun28 2023 build_openroad.sh --local --latest using Homebrew packages
```

### To Reproduce

run openroad without/with -threads 8, monitor # of threads used during drt

stefanottili, 2023-06-30:
```
read_lef ispd19_test6/ispd19_test6.input.lef.gz
read_def ispd19_test6/ispd19_test6.input.def.gz
set_thread_count 8
set_routing_layers -signal Metal2-Metal9 -clock Metal2-Metal9
set_global_routing_layer_adjustment Metal2-Metal9 0.5
global_route -verbose
#read_guides ispd19_test6/ispd19_test6.guide 
detailed_route -output_drc 5_route_drc.rpt -verbose 1
write_db route.db
exit
```

a) openroad ispd19_test6.or -log default
it seems that for "pin access" computation openroad uses 8 threads and then adds another 8 for detail route.
How can one tell whether all 2+16 processes are 'doing something' ?

b) openroad ispd19_test6.or -threads 8 -log threads
it only uses 2+8 threads total and overall runtime is faster.

rovinski, 2023-06-30:
Pin access and detail route are completely separate steps so even if the threads are created, only 8 of them should be active at a given time.

DRT in general uses OpenMP dynamic scheduling which I believe creates a thread pool and only destroys thread contexts when requested or on exit. I would be only mildly surprised if you see 8 threads being created on top of the 8 already in the thread pool, but I would be very surprised if all 16 threads are trying to be active at the same time. That's not what I see on my machine on the latest master.

A telltale sign that more threads than CPUs are active is if each thread's CPU utilization is <<~100% while running. Is that what you see?

stefanottili, 2023-06-30:
I noticed differences in runtime and number of threads reported.

a) running openroad with the set_thread_count 8 command leads to 16 threads and is slower then
b) specifying -threads 8 on the command line, which reports 8 threads in activity monitor.

Q1) why would using the same script without/with -threads 8 on the command line cause a different number of threads ?
Q2) why would that lead to runtime differences ?

I cannot tell wether all 16 threads are used during drt.

Disclaimer with regard to runtimes is that I've seen huge variations on MacBook Air M1 Ventura when running openroad multiple times without a reboot.

rovinski, 2023-07-01:
I'm not really able to replicate this on my machine, it would be helpful if you provided a log or some kind of data which showcases this behavior. It could be as with #3535 that this is a machine or environment-specific issue.

stefanottili, 2023-07-01:
There is nothing in the logs besides runtimes.
The “activity monitor” shows nearly 800% cpu usage, ~8gb of memory (of 16gb) and # of threads, either 10 or 18.

> I'm not really able to replicate this on my machine
How many #of threads do you see for a) and b) ?
Does ‘top’ give that info or what tool do you use on linux ?
 
I was hoping that some other apple silicon user would chime in.

stefanottili, 2023-07-02:
std::thread::hardware_concurrency() returns 8

maliberty, 2025-08-30:
I don't see a path forward as nobody has reproduced it on another platform.

closed 2025-08-30 by maliberty