The-OpenROAD-Project/OpenROAD #3563: repairDesign dumps core when buffering net connected to IO pad (output on core-side) and SRAM macro (input)

chaufe, 2023-07-03:
### Describe the bug

I have two similar test cases:
A non-clock net is connected to the core side output of an IO pad, and to an input of an SRAM macro. Additionally, the net has fan-out to a set of registers. I have two testcases for OpenROAD to dump core, one within the `resizer.tcl` script, one within the `gpl.tcl` script. Based on the logs, I assume both test cases to have the same root cause, so I'm filing this as one issue only.

**Testcase 1 (resizer):**
With only a few registers connected (two flops), the `resizer.tcl` script will dump core during `repairDesign`.

**Testcase 2 (gpl):**
With more registers connected (162 flops), even the `gpl.tcl` script will dump core during `repairDesign`.

**Workaround**
As a work-around, I have to manually add a buffer after the core-side output of the IO pad cell, and set the net between IO pad cell and buffer as `dont_touch`. OpenROAD will then create more buffers after output of the manually inserted buffer.

For both test cases, I have additionally attached an ipython notebook for Google Colab to reproduce the issue.

/cc @proppy 
/cc @dhaentz1

### Expected Behavior

OpenROAD should buffer the net connected to the core-side output of the IO pad.

### OpenROAD Environment

```shell
Env.sh seems not to be included in my litex-hub package - please use the testcase notebook to extract data.
```

### OpenLane Environment

```shell
env.py seems to be broken in my setup - please use the testcase notebook to extract data.
```

### To Reproduce

**Testcase 1 (resizer):**
[_build.coredump_resizer.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11934005/_build.coredump_resizer.zip)
[OpenRoad__coredump_when_resizing_net_driven_by_pad_and_connected_to_sram_macro.ipynb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11934022/OpenRoad__coredump_when_resizing_net_driven_by_pad_and_connected_to_sram_macro.ipynb.gz)

**Testcase 2 (gpl):**
[_build.coredump_gpl.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11934011/_build.coredump_gpl.zip)
[OpenRoad__coredump_when_global_placing_highfanout_net_driven_by_pad_and_connected_to_sram_macro.ipynb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11934027/OpenRoad__coredump_when_global_placing_highfanout_net_driven_by_pad_and_connected_to_sram_macro.ipynb.gz)

### Relevant log output

```shell
**Testcase 1 (resizer):**
[INFO]: Setting RC values...
[INFO RSZ-0058] Using max wire length 6781um.
Signal 11 received
Stack trace:
 0# 0x0000000000B9BB0B in openroad
 1# 0x00007F7E3FF54090 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getITerms() in openroad
 3# sta::dbNetwork::visitConnectedPins(sta::Net const*, sta::PinVisitor&, sta::NetSet&) const in openroad
 4# sta::Network::connectedPinIterator(sta::Net const*) const in openroad
 5# rsz::RepairDesign::hasInputPort(sta::Net const*) in openroad
 6# rsz::RepairDesign::makeRepeater(char const*, int, int, sta::LibertyCell*, bool, int, sta::Vector<sta::Pin const*>&, float&, float&, float&, sta::Net*&, sta::Pin*&, sta::Pin*&) in openroad
 7# rsz::RepairDesign::makeFanoutRepeater(sta::Vector<sta::Pin const*>&, sta::Vector<sta::Pin const*>&, odb::Rect, odb::Point, bool, bool, int, bool) in openroad
 8# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
 9# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
10# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
11# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
12# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
13# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
14# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
15# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
16# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
17# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
18# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
19# rsz::RepairDesign::repairNet(sta::Net*, sta::Pin const*, sta::Vertex*, bool, bool, bool, int, bool, int&, int&, int&, int&, int&) in openroad
20# rsz::RepairDesign::repairDesign(double, double, double, int&, int&, int&, int&, int&) in openroad
21# rsz::RepairDesign::repairDesign(double, double, double) in openroad
22# 0x0000000000CD5341 in openroad
23# TclNRRunCallbacks in /content/eda/bin/../lib/libtcl8.6.so
24# 0x00007F7E41A741BF in /content/eda/bin/../lib/libtcl8.6.so
25# Tcl_EvalEx in /content/eda/bin/../lib/libtcl8.6.so
26# Tcl_Eval in /content/eda/bin/../lib/libtcl8.6.so
27# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
28# ord::tclAppInit(Tcl_Interp*) in openroad
29# Tcl_MainEx in /content/eda/bin/../lib/libtcl8.6.so
30# main in openroad
31# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
32# 0x0000000000B85E22 in openroad

**Testcase 2 (gpl):**
[INFO]: Setting RC values...
[INFO RSZ-0058] Using max wire length 6781um.
Signal 11 received
Stack trace:
 0# 0x0000000000B9BB0B in openroad
 1# 0x00007F7E3FF54090 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getITerms() in openroad
 3# sta::dbNetwork::visitConnectedPins(sta::Net const*, sta::PinVisitor&, sta::NetSet&) const in openroad
 4# sta::Network::connectedPinIterator(sta::Net const*) const in openroad
 5# rsz::RepairDesign::hasInputPort(sta::Net const*) in openroad
 6# rsz::RepairDesign::makeRepeater(char const*, int, int, sta::LibertyCell*, bool, int, sta::Vector<sta::Pin const*>&, float&, float&, float&, sta::Net*&, sta::Pin*&, sta::Pin*&) in openroad
 7# rsz::RepairDesign::makeFanoutRepeater(sta::Vector<sta::Pin const*>&, sta::Vector<sta::Pin const*>&, odb::Rect, odb::Point, bool, bool, int, bool) in openroad
 8# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
 9# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
10# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
11# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
12# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
13# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
14# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
15# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
16# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
17# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
18# rsz::RepairDesign::makeRegionRepeaters(rsz::LoadRegion&, int, int, sta::Pin const*, bool, bool, int, bool) in openroad
19# rsz::RepairDesign::repairNet(sta::Net*, sta::Pin const*, sta::Vertex*, bool, bool, bool, int, bool, int&, int&, int&, int&, int&) in openroad
20# rsz::RepairDesign::repairDesign(double, double, double, int&, int&, int&, int&, int&) in openroad
21# rsz::RepairDesign::repairDesign(double, double, double) in openroad
22# 0x0000000000CD5341 in openroad
23# TclNRRunCallbacks in /content/eda/bin/../lib/libtcl8.6.so
24# 0x00007F7E41A741BF in /content/eda/bin/../lib/libtcl8.6.so
25# Tcl_EvalEx in /content/eda/bin/../lib/libtcl8.6.so
26# Tcl_Eval in /content/eda/bin/../lib/libtcl8.6.so
27# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
28# ord::tclAppInit(Tcl_Interp*) in openroad
29# Tcl_MainEx in /content/eda/bin/../lib/libtcl8.6.so
30# main in openroad
31# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
32# 0x0000000000B85E22 in openroad
```

openroadie, 2023-07-05:
@proppy I can reproduce the crash. It is happening when the repairDesign is trying to fix the rst_int net. I will debug more to figure out why but I was wondering if fanout fixing should be done on what seems to be the reset signal (the crash needs to be fixed regardless).

chaufe, 2023-07-05:
@openroadie Thanks for the heads-up. Fanout fixing needs to be done for any net. Otherwise transition times may increase outside of the characterized range of the libraries making timing analysis results inaccurate. It may also be a reliability issue as too-high fanouts may imply high switching currents on the driver pin possibly causing local voltage fluctuations as well as electromigration problems.
Here, `rst_int` is just an example for a synchronous input with multi-pin fanout.
(But I agree that a reset net would need proper constraining and/or circuitry to be useful.)

Not part of this test case but similar would be the shift-enable net for scan testing.
That is also a synchronous input net with high fanout (connected to all flops).

closed 2023-07-15 by maliberty

chaufe, 2023-07-20:
Thanks for the fix. The core-dumps are gone now.

PR #3579 (2023-07-15): fix  #3563: Handle top level port with no net connected to it
fixes #3563
commit 15e28bd8 (2023-07-15): Merge pull request #3579 from openroadie/issue_3563

fix  #3563: Handle top level port with no net connected to it
commit 840ceb89 (2023-07-15): fix  #3563: Handle top level port with no net connected to it