The-OpenROAD-Project/OpenROAD #3566: Problem with spef written after `read_def -child`

kareefardi, 2023-07-04:
### Describe the bug

After reading a macro with `read_def -child`, the design seems to be flattened. This is apparent in netlists generated using `write_verilog` and `write_def`. However the generated spef contains references to the flattened macro. Hence, that spef cannot be read again properly. 

### Expected Behavior

Be able to annotate a design that make use of `read_def -child` 

### Environment

```shell
Git commit: 914d6d23c442175889a2cca9b8e6fc37fe66e244
kernel: Linux 6.1.0-1015-oem
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9039-g914d6d23c
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 97 lines cut ...]
[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11947809/reproducible.tar.gz)

### Relevant log output

```shell
[WARNING STA-0174] message limit reached, this message will no longer print
[WARNING STA-0172] extract.spef line 2987, instance mprj not found.
[WARNING STA-0172] extract.spef line 3005, instance mprj not found.
[WARNING STA-0172] extract.spef line 3023, instance mprj not found.
[WARNING STA-0172] extract.spef line 3041, instance mprj not found.
[WARNING STA-0172] extract.spef line 3059, instance mprj not found.
[WARNING STA-0172] extract.spef line 3076, instance mprj not found.
[WARNING STA-0172] extract.spef line 3094, instance mprj not found.
[WARNING STA-0172] extract.spef line 3108, instance mprj not found.
[WARNING STA-0172] extract.spef line 3128, instance mprj not found.
[WARNING STA-0172] extract.spef line 3144, instance mprj not found.
[WARNING STA-0172] extract.spef line 3167, instance mprj not found.
[WARNING STA-0172] extract.spef line 3184, instance mprj not found.
[WARNING STA-0172] extract.spef line 3207, instance mprj not found.
[WARNING STA-0172] extract.spef line 3222, instance mprj not found.
[WARNING STA-0172] extract.spef line 3240, instance mprj not found.
[WARNING STA-0172] extract.spef line 3258, instance mprj not found.
[WARNING STA-0172] extract.spef line 3276, instance mprj not found.
[WARNING STA-0172] extract.spef line 3290, instance mprj not found.
```

maliberty, 2023-07-07:
What do you observe as different re:  "This is apparent in netlists generated using write_verilog and write_def."

read_def -child is mostly for display in the gui and doesn't flatten anything afaik.

kareefardi, 2023-07-09:
It appears that I was mistaken. Only write_verilog output has a full netlist. This is the netlist after reading the child macro. It has the inner macro components flattened. 
[result.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11995976/result.tar.gz)