The-OpenROAD-Project/OpenROAD #3569: get_pins dumps core if -filter is over-constrained

chaufe, 2023-07-04:
### Describe the bug

When over-constraining the `-filter` expression of `get_pins`, a core dump gets triggered.
To trigger the core-dump, I am using this example:
`get_pins -quiet -filter "direction == in && direction == out" *`

Other filter expression may also trigger a core dump.

I have additionally attached an ipython notebook for Google Colab to reproduce the issue.

/cc @proppy 
/cc @dhaentz1

### Expected Behavior

get_pins should return an empty list.

### OpenROAD Environment

```shell
Env.sh seems not to be included in my litex-hub package - please use the testcase notebook to extract data.
```

### OpenLane Environment

```shell
env.py seems to be broken in my setup - please use the testcase notebook to extract data.
```

### To Reproduce

[_build.coredump_get_pins.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11949228/_build.coredump_get_pins.zip)
[OpenRoad__coredump_upon_overconstrained_get_pins.ipynb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11949234/OpenRoad__coredump_upon_overconstrained_get_pins.ipynb.gz)

### Relevant log output

```shell
Skip floorplanning - just trigger core dump ...
Signal 11 received
Stack trace:
 0# 0x0000000000B9E2AB in openroad
 1# 0x00007FB63E9A1090 in /lib/x86_64-linux-gnu/libc.so.6
 2# filter_pins(char const*, char const*, char const*, sta::Vector<sta::Pin const*>*) in openroad
 3# 0x0000000000D6F97E in openroad
 4# TclNRRunCallbacks in /content/eda/bin/../lib/libtcl8.6.so
 5# 0x00007FB6404C11BF in /content/eda/bin/../lib/libtcl8.6.so
 6# Tcl_EvalEx in /content/eda/bin/../lib/libtcl8.6.so
 7# Tcl_Eval in /content/eda/bin/../lib/libtcl8.6.so
 8# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
 9# ord::tclAppInit(Tcl_Interp*) in openroad
10# Tcl_MainEx in /content/eda/bin/../lib/libtcl8.6.so
11# main in openroad
12# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
13# 0x0000000000B885C2 in openroad
```

openroadie, 2023-07-05:
@chaufe thank you for the test case. I can reproduce the problem and will work on getting a fix in ASAP.

chaufe, 2023-07-07:
Thanks for the fix. I can reproduce the issue is gone.

PR #3582 (2023-07-06): Issue 3569
Fixes #3569