The-OpenROAD-Project/OpenROAD #3570: LEF 5.7 property not supported

Blebowski, 2023-07-04:
### Describe the bug

I am trying to read in tech LEF file for an older proprietary PDK, and I obtain following messages:

```
[INFO ODB-0222] Reading LEF file: xyz_tech.lef
[INFO ODB-0388] unsupported LEF57_ARRAYSPACING property for layer VI1 :" ARRAYSPACING WIDTH 0.56 CUTSPACING 0.13 ARRAYCUTS 3 SPACING 0.13 ; "
[INFO ODB-0388] unsupported LEF57_ARRAYSPACING property for layer VI2 :" ARRAYSPACING WIDTH 0.56 CUTSPACING 0.13 ARRAYCUTS 3 SPACING 0.13 ; "
[INFO ODB-0388] unsupported LEF57_ARRAYSPACING property for layer VI3 :" ARRAYSPACING WIDTH 0.56 CUTSPACING 0.13 ARRAYCUTS 3 SPACING 0.13 ; "
[INFO ODB-0388] unsupported LEF57_ARRAYSPACING property for layer VI4 :" ARRAYSPACING WIDTH 0.56 CUTSPACING 0.13 ARRAYCUTS 3 SPACING 0.13 ; "
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 78 technology vias
[INFO ODB-0226] Finished LEF file:  xyz_tech.lef
```

### Expected Behavior

I would expect that:
1. This message is a warning (not an INFO)
2. Support of also older LEF/DEF versions (e.g. the mentioned 5.7)

### Environment

```shell
Git commit: 914d6d23c442175889a2cca9b8e6fc37fe66e244
kernel: Linux 4.18.0-425.10.1.el8_7.x86_64
os: Red Hat Enterprise Linux 8.7 (Ootpa)
cmake version 3.23.2
-- The CXX compiler identification is GNU 8.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9039-g914d6d23c
-- System name: Linux
-- Compiler: GNU 8.5.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /tools/tropic/ts-python-env/virtualenvs/ts-python-env-yQ9mF3ZL-py3.8/bin/python3.8 (found version "3.8.13") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
[... 103 lines cut ...]
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.75C7FiYRpS
```

### To Reproduce

Unfortunately, I can't provide the whole PDK lef.

Small extract from the techlef demonstrating the issue.

In property definitions section:

```
  LAYER LEF57_SPACING STRING ;
  LAYER LEF57_SPACINGTABLE STRING ;
  LAYER LEF57_ARRAYSPACING STRING ;
```
The definition of the LEF rule is in the message above.

maliberty, 2023-07-05:
@gadfort I suspect this world only affect pdn. Do you have any thoughts on support?

QuantamHD, 2023-07-05:
@maliberty Don't we already support the 5.8 version of this https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/src/lefin/ArraySpacingParser.cpp

If so, can't we just add a new parse handler for 5.7 and leverage the existing parser?

QuantamHD, 2023-07-05:
@Blebowski If you want to give it a shot I'm pretty sure if you just add another else if for this property with the existing parser it will "just work" ™.

gadfort, 2023-07-05:
@maliberty @QuantamHD I think we should be able to use the same underlying code with minimal changes. But first we need a parser.

Blebowski, 2023-07-07:
> 
Hi @QuantamHD , I will give it a try. Altough, I don't think it is a good idea trying to hack this "quick and dirty" way.
IMHO, this would deserve looking at the property differences between various LEF versions. Then, it may be
good to adjust property parsing so that different LEF versions are handled correctly for all properties (if that
is allowed by LEF spec). Let me see what I can do. Just a note, I don't really have much time lately, so it might
take some time to produce a patch.

Blebowski, 2023-07-07:
OK, this is weird. I am trying to search for `LEF57_` prefixed properties in LEF 5.7 RM, however, I see only `LEF58_` prefixed ones. The LEF file itself states that it uses LEF version 5.5, however, I am unable to find 5.5 RM. I am probably missing something, or properties with `LEF57_` prefix is something that is not really according to LEF/DEF RM. Anyone has a clue
where this is coming from?

maliberty, 2023-07-07:
I don't think there is any such property as LEF57_ARRAYSPACING.   Even if you look at an older 5.7 manual like https://www.ispd.cc/contests/18/lefdefref.pdf you don't find it (just ARRAYSPACING).  Where did this LEF come from?

Blebowski, 2023-07-07:
@maliberty, the file header states following:
```
LEF file generated by Abstract Generator version 5.5.9
```

maliberty, 2023-07-07:
What tool is that?  From what I can see it writes invalid LEF.

Blebowski, 2023-07-07:
I don't know, but apparently it is a tool from Cadence, e.g.:

https://redirect.cs.umbc.edu/~tinoosh/cmpe641/slides/lect03_abstract.pdf

https://www.youtube.com/watch?v=c0IgLac4guI

maliberty, 2023-07-07:
With no document defining this property it seem a bit odd to "support" it.  Are you using an older version of Virtuoso?

Blebowski, 2023-07-07:
Yup, I agree. I did not generate the LEF file, we have it delivered from fab. The date says 2014, so if it is a bug, it is possible that newer version of Virtuoso will have the issue fixed. I need to ask our fab if they can provide more up to date version.

maliberty, 2023-07-07:
Worst case I would make a local copy and just change it to LEF58 and see if that works.

Blebowski, 2023-07-08:
OK, I will give it a try.

Blebowski, 2023-07-19:
After trying the "hack" to replace the `LEF57_` with `LEF58_` the LEF file loads in correctly, so I will close the issue.

closed 2023-07-19 by Blebowski