The-OpenROAD-Project/OpenROAD #3571: make_port dumps core

chaufe, 2023-07-04:
### Describe the bug

Calling `make_ports dummy_port input` triggers a core-dump.

I have additionally attached an ipython notebook for Google Colab to reproduce the issue.

/cc @proppy 
/cc @dhaentz1

### Expected Behavior

`make_ports` should not core-dump but trigger an error message if required.

### OpenROAD Environment

```shell
Env.sh seems not to be included in my litex-hub package - please use the testcase notebook to extract data.
```

### OpenLane Environment

```shell
env.py seems to be broken in my setup - please use the testcase notebook to extract data.
```

### To Reproduce

[_build.core_dump__make_port.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11949738/_build.core_dump__make_port.zip)
[OpenRoad__make_port_dumps_core.ipynb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11949743/OpenRoad__make_port_dumps_core.ipynb.gz)

### Relevant log output

```shell
Create a port ...
Signal 11 received
Stack trace:
 0# 0x0000000000B9E2AB in openroad
 1# 0x00007FD1088FB090 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::ConcreteInstance::addPin(sta::ConcretePin*) in openroad
 3# sta::ConcreteNetwork::makePin(sta::Instance*, sta::Port*, sta::Net*) in openroad
 4# sta::Sta::makePortPin(char const*, char const*) in openroad
 5# 0x0000000000D5F7FB in openroad
 6# TclNRRunCallbacks in /content/eda/bin/../lib/libtcl8.6.so
 7# 0x00007FD10A41B1BF in /content/eda/bin/../lib/libtcl8.6.so
 8# Tcl_EvalEx in /content/eda/bin/../lib/libtcl8.6.so
 9# Tcl_Eval in /content/eda/bin/../lib/libtcl8.6.so
10# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
11# ord::tclAppInit(Tcl_Interp*) in openroad
12# Tcl_MainEx in /content/eda/bin/../lib/libtcl8.6.so
13# main in openroad
14# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
15# 0x0000000000B885C2 in openroad
```

openroadie, 2023-07-07:
I am little confused about this issue. If STA is integrated into Openroad and has callbacks to stay in sync with the Openroad DB (odb) ... how is it possible to create a port via what seems to be a pure STA command (since I don't think the call back mechanism is bi-directional). 

cc: @tspyrou @maliberty

maliberty, 2023-07-07:
dbNetwork::connect makes a bterm (called from dbSta::connectPin) so it seems bidirectional.

openroadie, 2023-07-07:
Thanks. Let me take a look at that.

commit 39de38a3 (2023-10-26): dbSta: handle make_port

Fixes #3571
closed 2023-10-26 by maliberty