The-OpenROAD-Project/OpenROAD #3583: Save DRC reports intermittently for global routing for hard runs

oharboe, 2023-07-05:
### Description

If global routing is having problmes converging, it can run for many hours.

However, even after a few minutes, a DRC report could contain valuable information about where the routing problems are.

### Suggested Solution

Save out DRC reports "occasionally". This should be in some idempotent interval, i.e. wall clock time or type of machine that global routing is running on should not matter.

Furthermore, this should be the default behavior. Spelunking in .tcl scripts shouldn't be required to get information on why global routing is struggling.

maliberty, 2023-07-07:
In general it isn't safe to issue a message from a signal handler (see https://stackoverflow.com/questions/16891019/how-to-avoid-using-printf-in-a-signal-handler).  It is also non-standard that ctrl-c make run another 10+ minutes.  I know of no other programs that do so.

I think it better to just write the report on each iteration.  We could have some upper limit on the number of violations to prevent it from generating huge reports on the early iterations (say 100k).

QuantamHD, 2023-07-07:
I think that's acceptable @maliberty, but dd is an example where you can interrupt the program for a status report. https://askubuntu.com/a/215521

maliberty, 2023-07-07:
dd gives an immediate result - I've seen that in other programs.  This would be quite different where the response occurs after an unbounded amount of time.

oharboe, 2023-07-07:
> dd gives an immediate result - I've seen that in other programs. This would be quite different where the response occurs after an unbounded amount of time.
> 
> It also not a status report - it still eventually exits the program.

The approach may require some care and not be entirely typical, but it is not unprecedented. I'm fine with any solution to this reporting problem where I don't have to configure the build or go spelunking in .tcl.

I haven't checked, but I asked chatgpt...

MySQL Server: MySQL Server is a highly popular relational database management system. When you send a SIGINT signal to MySQL, it doesn't immediately terminate. It attempts to finish active transactions, roll back uncommitted transactions, and clean up the system, in order to maintain data consistency before shutting down.

Apache HTTP Server: As a popular web server application, Apache has to handle many active connections simultaneously. When a SIGINT signal is sent, it tries to gracefully stop by allowing active connections to complete while not accepting new connections.

TensorFlow: This machine learning library used for high performance numerical computation can run processes that take a significant amount of time, such as training neural networks. When interrupted, it might need to save the state of the model before exiting.

Oracle Database: Much like MySQL, when an Oracle Database server process receives a SIGINT, it goes through a sequence of steps to ensure data consistency and integrity. This includes committing or rolling back transactions and writing necessary logs.

maliberty, 2023-07-08:
I think dumping out a report per iteration, or some multiple, is fine.  I feel this is heading in the wrong direction.  It will only work interactively, only in drt, and differently from every other command in OR.  It will be obscure as well.

QuantamHD, 2023-07-08:
Agreed Matt I think dumping the DRC report with an option up front would be good. I think we could lean on the GUI for more interactive results

oharboe, 2023-07-12:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/3595 merged.

Now the question remains as to how to support this in ORFS In a way where potential benefactors will actually think of and find out how to use this feature.

Ideas:

- Verbose feature in ORFS? `make VERBOSE=1`?
- Hack GLOBAL_ROUTING_ARGS? The user would have to find this option, find out what current GLOBAL_ROUTING_ARGS and add this argument, temporarily. What about detailed routing, where progress .rpt files would be similarly useful? Hacking detailed routing args is even harder than GLOBAL_ROUTING_ARGS.

maliberty, 2023-07-12:
#3595 was a change in detailed routing not global routing.  I think you are mixing issues.

oharboe, 2023-07-12:
> #3595 was a change in detailed routing not global routing. I think you are mixing issues.

I am, but the same usability concern applies.

maliberty, 2023-07-12:
Ok but they are different tasks that will likely be handled by different people so let's not mix the two together.

oharboe, 2023-07-12:
> Ok but they are different tasks that will likely be handled by different people so let's not mix the two together.

Sure, but a consistent user experience is desireable.

maliberty, 2023-10-17:
done

closed 2023-10-17 by maliberty