The-OpenROAD-Project/OpenROAD #3587: GUI module separate and could not pull back

linzihao123456, 2023-07-07:
### Describe the bug

I just clicked the separate button of the scripting module of the GUI. And I could not pull it back. In this case, the scripting module could not work. The close button and the separate button don't work. I guess the other modules, like the display control, also have the same bug. 
I have tried to rebuild the project and even deleted the project and cloned it again, the GUI seem to have the same configuration. The modules that have been separated stay still. 

[image]

### Expected Behavior

The separated modules can be embedded into the GUI.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
[ 11%] Creating directories for 'googletest-populate'
[ 22%] Performing download step (download, verify and extract) for 'googletest-populate'
-- Downloading...
   dst='/tmp/tmp.8pQWFCoHdw/_deps/googletest-subbuild/googletest-populate-prefix/src/v1.13.0.zip'
   timeout='none'
   inactivity timeout='none'
-- Using src='https://github.com/google/googletest/archive/refs/tags/v1.13.0.zip'
CMake Error at googletest-subbuild/googletest-populate-prefix/src/googletest-populate-stamp/download-googletest-populate.cmake:170 (message):
  Each download failed!

    error: downloading 'https://github.com/google/googletest/archive/refs/tags/v1.13.0.zip' failed
          status_code: 92
          status_string: "Stream error in the HTTP/2 framing layer"
          log:
          --- LOG BEGIN ---
            Trying 20.205.243.166:443...

  Connected to github.com (20.205.243.166) port 443 (#0)

  ALPN: offers h2

  ALPN: offers http/1.1

  [5 bytes data]

  TLSv1.3 (OUT), TLS handshake, Client hello (1):

  [512 bytes data]

  [5 bytes data]

  TLSv1.3 (IN), TLS handshake, Server hello (2):

  [122 bytes data]

  [5 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):

  [19 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, Certificate (11):

  [2459 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, CERT verify (15):

  [78 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, Finished (20):

  [36 bytes data]

  [5 bytes data]

  TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):

  [1 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (OUT), TLS handshake, Finished (20):

  [36 bytes data]

  SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256

  ALPN: server accepted h2

  Server certificate:

   subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
   start date: Feb 14 00:00:00 2023 GMT
   expire date: Mar 14 23:59:59 2024 GMT
   subjectAltName: host "github.com" matched cert's "github.com"
   issuer: C=US; O=DigiCert Inc; CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1
   SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.

  Using HTTP2, server supports multiplexing

  Copying HTTP/2 data in stream buffer to connection buffer after upgrade:
  len=0

  [5 bytes data]

  [1 bytes data]

  [5 bytes data]

  [1 bytes data]

  [5 bytes data]

  [1 bytes data]

  h2h3 [:method: GET]

  h2h3 [:path: /google/googletest/archive/refs/tags/v1.13.0.zip]

  h2h3 [:scheme: https]

  h2h3 [:authority: github.com]

  h2h3 [user-agent: curl/7.83.1]

  h2h3 [accept: */*]

  Using Stream ID: 1 (easy handle 0x556ad70395b0)

  [5 bytes data]

  [1 bytes data]

  GET /google/googletest/archive/refs/tags/v1.13.0.zip HTTP/2

  Host: github.com

  user-agent: curl/7.83.1

  accept: */*

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):

  [57 bytes data]

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):

  [57 bytes data]

  old SSL session ID is stale, removing

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (IN), TLS alert, close notify (256):

  [2 bytes data]

  HTTP/2 stream 1 was not closed cleanly before end of the underlying stream

  [5 bytes data]

  [1 bytes data]

  Closing connection 0

  [5 bytes data]

  [1 bytes data]

  TLSv1.3 (OUT), TLS alert, close notify (256):

  [2 bytes data]

          --- LOG END ---

make[2]: *** [CMakeFiles/googletest-populate.dir/build.make:100: googletest-populate-prefix/src/googletest-populate-stamp/googletest-populate-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/googletest-populate.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

CMake Error at /usr/local/share/cmake-3.24/Modules/FetchContent.cmake:1604 (message):
  Build step for googletest failed: 2
Call Stack (most recent call first):
  /usr/local/share/cmake-3.24/Modules/FetchContent.cmake:1744:EVAL:2 (__FetchContent_directPopulate)
  /usr/local/share/cmake-3.24/Modules/FetchContent.cmake:1744 (cmake_language)
  /usr/local/share/cmake-3.24/Modules/FetchContent.cmake:1958 (FetchContent_Populate)
  third-party/gtest/CMakeLists.txt:8 (FetchContent_MakeAvailable)

-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9036-g12e051bf4
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Configuring incomplete, errors occurred!
```

### To Reproduce

May no error here. Because it can run the whole flow. 

### Screenshots

[image]
 The same as the display control module. They could not pull back and work.

maliberty, 2023-07-07:
Have tried clicking the leftmost of the buttons in the upper right of the window?  It should re-dock.
[image]

You can also removing ~/.config/OpenRoad Project/ on linux to reset the window between runs.

maliberty, 2023-10-17:
no response

closed 2023-10-17 by maliberty