The-OpenROAD-Project/OpenROAD #3592: mock-array detailed routing violations

oharboe, 2023-07-08:
### Describe the bug

To reproduce, use https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1210, create `settings.mk`

```
export DESIGN_CONFIG?=designs/asap7/mock-array/config.mk
export MOCK_ARRAY_DATAWIDTH?=8
export MOCK_ARRAY_TABLE?=8 8 4 4 5 5
export MOCK_ARRAY_SCALE?=320
```

Run `make verilog` and then `make`

[image]

[image]

### Expected Behavior

Detailed route errors don't happen with smaller multiples of MOCK_ARRAY_SCALE.

### Environment

```shell
lastet master
```

### To Reproduce

See above.

oharboe, 2023-07-08:
Also happens with:

```
export DESIGN_CONFIG?=designs/asap7/mock-array/config.mk
export MOCK_ARRAY_DATAWIDTH?=8
export MOCK_ARRAY_TABLE?=8 8 4 4 5 5
export MOCK_ARRAY_SCALE?=80
```

stripped down `make detail_route_issue` 
[detail_route_mock-array_asap7_8-8-80-8-8-4-4-5-5_2023-07-08_22-04.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11992332/detail_route_mock-array_asap7_8-8-80-8-8-4-4-5-5_2023-07-08_22-04.tar.gz)

maliberty, 2023-07-08:
Are you sure the pins are aligned?

oharboe, 2023-07-08:
> Are you sure the pins are aligned?

I dont know how to check... thought positioning macros on mutiples of 0.048 would take care of it...

Could you provide some "pin alignment check for dummies" instructions?

If you teach me how to fish, I should be filing fewer.bogus ssues....

oharboe, 2023-07-08:
@maliberty I believe that this is showing that the pins are misaligned by 1nm. That's not intentional, the pins should be on multiples of 0.048, rounding error somewhere?

[image]

maliberty, 2023-07-08:
It's really best to work in integer dbu not microns. I expect it related to your script changes

oharboe, 2023-07-09:
> It's really best to work in integer dbu not microns. I expect it related to your script changes

Agreed. I can't get to the `set units [$block getDefUnits]` API from my Python code though.

oharboe, 2023-07-09:
Should be fixed by https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1212, will re-open if needed.

closed 2023-07-09 by oharboe