The-OpenROAD-Project/OpenROAD #3609: Crash on "Update" timing

oharboe, 2023-07-12:
### Describe the bug

```
Loading spef
Critical: /OpenROAD-flow-scripts/tools/OpenROAD/src/sta/search/PathEnum.cc line 574, diversion path not found
Signal 11 received
Stack trace:
Signal 11 received
Stack trace:
 0# 0x0000000000D4AE07 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FE06C808400 in /lib64/libc.so.6
 2# QPen::QPen(Qt::PenStyle) in /lib64/libQt5Gui.so.5
 3# QPaintEngineExPrivate::QPaintEngineExPrivate() in /lib64/libQt5Gui.so.5
 4# 0x00007FE06E05F519 in /lib64/libQt5Gui.so.5
 5# QRasterPaintEngine::QRasterPaintEngine(QPaintDevice*) in /lib64/libQt5Gui.so.5
 6# QImage::paintEngine() const in /lib64/libQt5Gui.so.5
 7# QPainter::begin(QPaintDevice*) in /lib64/libQt5Gui.so.5
 8# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# gui::RenderThread::run() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# 0x00007FE06D98EE71 in /lib64/libQt5Core.so.5
11# 0x00007FE07128DEA5 in /lib64/libpthread.so.0
12# clone in /lib64/libc.so.6
make: *** [gui_6_final.odb] Segmentation fault (core dumped)
```

### Expected Behavior

no crash

### Environment

```shell
latest master
```

### To Reproduce

1. load design. I can't share what crashes right now, but will look into providing a test case.
2. click "Update" on timing

Segfault above.

oharboe, 2023-07-12:
@maliberty Can I use deltaDebug.py to create an example?

maliberty, 2023-07-12:
I dont think there is a tcl equivalent to the update button (@gadfort ?)

maliberty, 2023-07-12:
This is a totally different stack (no sta) and an odd one that is inside of Qt.  I have no guesses

oharboe, 2023-07-12:
> This is a totally different stack (no sta) and an odd one that is inside of Qt. I have no guesses

Ouch. Any thoughts on how to reproduce a test-case? Do a meeting to debug?

gadfort, 2023-07-12:
It looks like the Qt stack trace is related to the fact that STA is crashing

maliberty, 2023-07-12:
@gadfort would it be easy to hack a way just to enable deltaDebug - it doesn't need to be good enough to commit.

gadfort, 2023-07-12:
@maliberty I suspect it is, you just need to add a tcl swig thing and associated functions needed to call the update button in the timing widget

oharboe, 2023-07-12:
@maliberty Some new tidbits... I have been using the `set_dont_touch` flag. When I tried #3611, I got the segfault below, but I was able to move on by disabling this code:

```
$ git diff
diff --git a/src/odb/src/db/dbITerm.cpp b/src/odb/src/db/dbITerm.cpp
index 92365272e..087c976e3 100644
--- a/src/odb/src/db/dbITerm.cpp
+++ b/src/odb/src/db/dbITerm.cpp
@@ -526,23 +526,23 @@ void dbITerm::disconnect()
     return;
 
   _dbInst* inst = iterm->getInst();
-  if (inst->_flags._dont_touch) {
-    inst->getLogger()->error(
-        utl::ODB,
-        370,
-        "Attempt to disconnect iterm of dont_touch instance {}",
-        inst->_name);
-  }
+  // if (inst->_flags._dont_touch) {
+  //   inst->getLogger()->error(
+  //       utl::ODB,
+  //       370,
+  //       "Attempt to disconnect iterm of dont_touch instance {}",
+  //       inst->_name);
+  // }
 
   _dbBlock* block = (_dbBlock*) iterm->getOwner();
   _dbNet* net = block->_net_tbl->getPtr(iterm->_net);
 
-  if (net->_flags._dont_touch) {
-    inst->getLogger()->error(utl::ODB,
-                             372,
-                             "Attempt to disconnect iterm of dont_touch net {}",
-                             net->_name);
-  }
+  // if (net->_flags._dont_touch) {
+  //   inst->getLogger()->error(utl::ODB,
+  //                            372,
+  //                            "Attempt to disconnect iterm of dont_touch net {}",
+  //                            net->_name);
+  // }
 
   for (auto callback : block->_callbacks)
     callback->inDbITermPreDisconnect(this);
```

```
___________________________________
Reading base odb file
Insts level debugging
Number of Insts 344095
Number of Nets 61976
Number of elements to be cut is 172047, [C#]
Signal 11 received
Stack trace:
 0# 0x000056018FE9FA4B in openroad
 1# 0x00007FD64F642520 in /lib/x86_64-linux-gnu/libc.so.6
 2# void utl::Logger::log<char*>(utl::ToolId, spdlog::level::level_enum, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char* const&) in openroad
 3# void utl::Logger::error<char*>(utl::ToolId, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char* const&) in openroad
 4# odb::dbITerm::disconnect() in openroad
 5# odb::dbInst::destroy(odb::dbInst*) in openroad
 6# 0x0000560191847F6F in openroad
 7# 0x00007FD64FF2B6B8 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 8# _PyObject_Call in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 9# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
10# 0x00007FD64FFC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
11# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
12# 0x00007FD64FFC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
13# 0x00007FD64FEE7618 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
14# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
15# 0x00007FD64FFC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
16# 0x00007FD64FEE7618 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
17# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
18# 0x00007FD64FFC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
19# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
20# 0x00007FD64FFC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
21# PyEval_EvalCode in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
22# 0x00007FD6500107BD in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
23# _PyRun_SimpleFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
24# _PyRun_AnyFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
25# Py_RunMain in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
26# Py_Main in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
27# main in openroad
28# 0x00007FD64F629D90 in /lib/x86_64-linux-gnu/libc.so.6
29# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
30# _start in openroad
74924 Segmentation fault      (core dumped) "$@"
```

oharboe, 2023-07-12:
@maliberty What is the Tcl command to quit the GUI? In deltaDebug.py I have to rely on timeout for now. I could speed it up quite a bit if I could add a quit command in tcl.

oharboe, 2023-07-12:
> @maliberty What is the Tcl command to quit the GUI? In deltaDebug.py I have to rely on timeout for now. I could speed it up quite a bit if I could add a quit command in tcl.

`exit` :-)

oharboe, 2023-07-12:
More crashes in deltaDebug:
```
Reading base odb file
Nets level debugging
Number of Insts 12
Number of Nets 61976
Number of elements to be cut is 30988, [C#]
Signal 11 received
Stack trace:
 0# 0x000055C47D2D49FB in openroad
 1# 0x00007F16D3242520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::destroy(odb::dbNet*) in openroad
 3# 0x000055C47EC7CB8F in openroad
 4# 0x00007F16D3B2B6B8 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 5# _PyObject_Call in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 6# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 7# 0x00007F16D3BC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 8# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
 9# 0x00007F16D3BC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
10# 0x00007F16D3AE7618 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
11# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
12# 0x00007F16D3BC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
13# 0x00007F16D3AE7618 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
14# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
15# 0x00007F16D3BC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
16# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
17# 0x00007F16D3BC681F in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
18# PyEval_EvalCode in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
19# 0x00007F16D3C107BD in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
20# _PyRun_SimpleFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
21# _PyRun_AnyFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
22# Py_RunMain in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
23# Py_Main in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
24# main in openroad
25# 0x00007F16D3229D90 in /lib/x86_64-linux-gnu/libc.so.6
26# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
27# _start in openroad
```

oharboe, 2023-07-12:
Avoiding crash above:

```
$ git diff
diff --git a/src/odb/src/db/dbITerm.cpp b/src/odb/src/db/dbITerm.cpp
index 92365272e..087c976e3 100644
--- a/src/odb/src/db/dbITerm.cpp
+++ b/src/odb/src/db/dbITerm.cpp
@@ -526,23 +526,23 @@ void dbITerm::disconnect()
     return;
 
   _dbInst* inst = iterm->getInst();
-  if (inst->_flags._dont_touch) {
-    inst->getLogger()->error(
-        utl::ODB,
-        370,
-        "Attempt to disconnect iterm of dont_touch instance {}",
-        inst->_name);
-  }
+  // if (inst->_flags._dont_touch) {
+  //   inst->getLogger()->error(
+  //       utl::ODB,
+  //       370,
+  //       "Attempt to disconnect iterm of dont_touch instance {}",
+  //       inst->_name);
+  // }
 
   _dbBlock* block = (_dbBlock*) iterm->getOwner();
   _dbNet* net = block->_net_tbl->getPtr(iterm->_net);
 
-  if (net->_flags._dont_touch) {
-    inst->getLogger()->error(utl::ODB,
-                             372,
-                             "Attempt to disconnect iterm of dont_touch net {}",
-                             net->_name);
-  }
+  // if (net->_flags._dont_touch) {
+  //   inst->getLogger()->error(utl::ODB,
+  //                            372,
+  //                            "Attempt to disconnect iterm of dont_touch net {}",
+  //                            net->_name);
+  // }
 
   for (auto callback : block->_callbacks)
     callback->inDbITermPreDisconnect(this);
diff --git a/src/odb/src/db/dbNet.cpp b/src/odb/src/db/dbNet.cpp
index 45030a3a5..73dbe6c81 100644
--- a/src/odb/src/db/dbNet.cpp
+++ b/src/odb/src/db/dbNet.cpp
@@ -2941,8 +2941,8 @@ void dbNet::destroy(dbNet* net_)
   _dbBlock* block = (_dbBlock*) net->getOwner();
 
   if (net->_flags._dont_touch) {
-    net->getLogger()->error(
-        utl::ODB, 364, "Attempt to destroy dont_touch net {}", net->_name);
+    // net->getLogger()->error(
+    //     utl::ODB, 364, "Attempt to destroy dont_touch net {}", net->_name);
   }
 
   dbSet<dbITerm> iterms = net_->getITerms();
@@ -2979,12 +2979,12 @@ void dbNet::destroy(dbNet* net_)
   }
 
   if (block->_journal) {
-    debugPrint(block->getImpl()->getLogger(),
-               utl::ODB,
-               "DB_ECO",
-               1,
-               "ECO: destroy net, id: {}",
-               net->getId());
+    // debugPrint(block->getImpl()->getLogger(),
+    //            utl::ODB,
+    //            "DB_ECO",
+    //            1,
+    //            "ECO: destroy net, id: {}",
+    //            net->getId());
     block->_journal->beginAction(dbJournal::DELETE_OBJECT);
     block->_journal->pushParam(dbNetObj);
     block->_journal->pushParam(net->getId());
```

maliberty, 2023-07-13:
Delta debug just needs to be taught to skip don't touch objects when deleting

maliberty, 2023-10-17:
@oharboe is this still relevant?

oharboe, 2023-10-18:
> @oharboe is this still relevant?

Fixed this summer. Thanks!

closed 2023-10-18 by rovinski