The-OpenROAD-Project/OpenROAD #3615: can't read "warn_error": no such variable error

gadfort, 2023-07-13:
### Describe the bug

When calling `set_dont_use` it's erroring our with the following message:
`can't read "warn_error": no such variable`

This may relate to https://github.com/The-OpenROAD-Project/OpenSTA/blob/7507ce1c51c76e868dba253d8579e34c7195ca2d/tcl/CmdArgs.tcl#L608

### Expected Behavior

Does not error out on unknown variable, but instead generate a proper error message or ignore the likely missing cell.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.26.4
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9148-gb28bd5735
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 95 lines cut ...]

### To Reproduce

[sc_issue_gcd_job0_skywater130_demo_orfs_floorplan0_20230713-165321.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12044414/sc_issue_gcd_job0_skywater130_demo_orfs_floorplan0_20230713-165321.tar.gz)

```
tar xvf sc_issue_gcd_job0_skywater130_demo_orfs_floorplan0_20230713-165321.tar.gz
cd sc_issue_gcd_job0_skywater130_demo_orfs_floorplan0_20230713-165321
./run.sh
```

### Relevant log output

```shell
[INFO ODB-0223]     Created 11 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  /home/pgadfort/siliconcompiler/third_party/pdks/skywater/skywater130/pdk/v0_0_2/apr/sky130_fd_sc_hd.tlef
Reading lef: /home/pgadfort/siliconcompiler/third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0222] Reading LEF file: /home/pgadfort/siliconcompiler/third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0225]     Created 437 library cells
[INFO ODB-0226] Finished LEF file:  /home/pgadfort/siliconcompiler/third_party/pdks/skywater/skywater130/libs/sky130hd/v0_0_2/lef/sky130_fd_sc_hd_merged.lef
Reading netlist verilog: inputs/gcd.vg
Reading SDC: /home/pgadfort/scgallery/scgallery/designs/gcd/constraints/sky130hd.sdc
Error: sc_apr.tcl, 347 can't read "warn_error": no such variable
```

openroadie, 2023-07-14:
@gadfort Looks like a malformed list that is being passed into get_lib_cell_args. 

in sc_apr.tcl: 347

If I do:
puts $sc_dontuse
**{sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8} sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 ...sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4 # Missing a closing brace here**

If I manually re-specify the list with all the cells right before:

set_dont_use $sc_dontuse

Then this error does not occur any more. I think this issue can be closed if you agree.

gadfort, 2023-07-14:
@openroadie I see the issue, but the error stems from an error in the evaluation. So I can fix the path, but the error is still there. It should be fixed in OpenSTA so it correctly fails with an error message

gadfort, 2023-07-26:
@openroadie FYI

maliberty, 2023-07-26:
@vvbandeira FYI

vvbandeira, 2023-07-26:
@gadfort, which commit of OpenROAD and OpenSTA are you using?

gadfort, 2023-07-26:
v2.0-9504-gd43619e08 <- this has some edits beyond the current head of master

vvbandeira, 2023-07-27:
@gadfort, I have compiled this version and run `./src/rsz/test/regression set_dontuse1` which passed, the log shows: 
```
[WARNING STA-0305] object 'non_existent_cell' not found.
```
Could you share the contents of `$sc_dontuse`? Maybe this is another corner case.

gadfort, 2023-07-27:
@vvbandeira I don't see that test in: https://github.com/The-OpenROAD-Project/OpenROAD/tree/2313c4895460a815bf1bce666ded4a52e34074f2/src/rsz/test is it possible it's local to your setup only? Do you mean set_dont_touch1?
I just recompiled, if I switch the line in set_dont_touch1 from `set_dont_touch` to `set_dontuse` I can trigger it still.

vvbandeira, 2023-07-27:
@gadfort, apologies for the confusion I mistyped, `set_dont_touch1` is the correct test.
The commit and `rsz` test I am referring to are these: `d43619e08` and test [link](https://github.com/gadfort/OpenROAD/blob/d43619e08c7d106e016d86249666e8f531b5727a/src/rsz/test/set_dont_touch1.tcl) from your fork.

---

Could you explain in more detail this:
> I just recompiled, if I switch the line in set_dont_touch1 from set_dont_touch to set_dontuse I can trigger it still.

I get `Error: set_dont_touch1.tcl, 8 invalid command name "set_dontuse"`. I don't think `set_dontuse` is a valid command -- if this was because of my typo, please ignore.

---

The `$sc_dontuse` contents I was asking for was from [this post](https://github.com/The-OpenROAD-Project/OpenROAD/issues/3615#issuecomment-1652114956) where you run:
> openroad> set_dont_use $sc_dontuse

Maybe the contents of `$sc_dontuse` have something special that triggers the error? I still cannot reproduce the error on my side. Do you have a test case or minimal example I can run on my side?

gadfort, 2023-07-27:
@vvbandeira the command is `set_dont_use` I missed the `_` when typing it out.
There is a typo the glob for one of the cells in `$sc_dontuse`.

This triggers it for me:
```
# set_dont_use on cell that does not exists

source "helpers.tcl"

read_liberty Nangate45/Nangate45_typ.lib
read_lef Nangate45/Nangate45.lef
read_def "gcd_nangate45_placed.def"
set_dont_use "non_existent_cell"
```

PR #3620 (2023-07-15): fix for OR #3615
fixes #3615 

cc: @gadfort
PR #3765 (2023-08-02): rsz: fix set_dont_use if cell is misspelled
Closes #3615