The-OpenROAD-Project/OpenROAD #3616: support removing dont_touch in deltaDebug.py

oharboe, 2023-07-14:
### Description

The `dont_touch` nets can contain significant confidential information.

### Suggested Solution

Although deltaDebug.py will work correctly on designs that have `dont_touch` after https://github.com/The-OpenROAD-Project/OpenROAD/pull/3614, it would be better if these nets can be removed entirely to whittle away confidential information.

maliberty, 2023-07-14:
Some problems might require the dont touch in order to reproduce.  We could have an option to remove the dont touch status.

oharboe, 2023-07-14:
> Some problems might require the dont touch in order to reproduce. We could have an option to remove the dont touch status.

Default should be to remove as much as possible, I think.

I think the primary concern of deltaDebug.py is confidentiality. Certainly there are size optimizations that are not done and make foo_issue includes.pretty much everything...

maliberty, 2023-07-14:
The primary concern is to reproduce the problem first.  The script was written for reducing problems for faster debug.  Confidentiality wasn't even a consideration at that time.

oharboe, 2023-07-14:
> The primary concern is to reproduce the problem first. The script was written for reducing problems for faster debug. Confidentiality wasn't even a consideration at that time.

I see. So to support this workflow for confidentiality purposes is indeed a feature request.

maliberty, 2023-07-15:
Making the design smaller serves both purposes though.  I'll send a PR for you test and see if it can reduce further.

maliberty, 2023-07-15:
@oharboe please try https://github.com/The-OpenROAD-Project/OpenROAD/pull/3623 and see if it works better at reducing your test case

oharboe, 2023-07-15:
> @oharboe please try #3623 and see if it works better at reducing your test case

Seems to work well :-)

closed 2023-07-15 by maliberty