The-OpenROAD-Project/OpenROAD #3624: dbSta compile error building OpenROAD locally

Iyury1, 2023-07-15:
### Describe the bug

I am building OpenROAD with these commands:

```
mkdir build && cd build
cmake ..
cmake --build . -j 8
```

I receive the following error:
```
/home/ian/OpenROAD/src/dbSta/src/dbSta.cc: In member function ‘virtual void sta::dbStaCbk::inDbBTermCreate(odb::dbBTerm*)’:
/home/ian/OpenROAD/src/dbSta/src/dbSta.cc:579:9: error: ‘class sta::dbSta’ has no member named ‘makePortPinAfter’
  579 |   sta_->makePortPinAfter(pin);
      |         ^~~~~~~~~~~~~~~~
```

### Expected Behavior

Compilation succeeds without error.

### Environment

```shell
Git commit: 8619564c5b7569ab6742cd1be8ba74d7e9863f0d
kernel: Linux 5.15.0-76-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9222-g8619564c5
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.9 (found version "3.9.5") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 99 lines cut ...]
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.MPFdzTlAcW
```

### To Reproduce

```
mkdir build && cd build
cmake ..
cmake --build . -j 8
```
or
```
make
```

maliberty, 2023-07-15:
Most likely your sta submodule is out of date.  Try ```git submodule update --init --recursive```

Iyury1, 2023-07-15:
That was it, thanks!

closed 2023-07-15 by Iyury1