The-OpenROAD-Project/OpenROAD #3634: Detailed routing can create tiny loops once in a blue moon

oharboe, 2023-07-17:
@osamahammad21 any thoughts on the loop?

_Originally posted by @maliberty in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3597#discussioncomment-6398612_

osamahammad21, 2023-07-21:
@oharboe   Can you post the full environment to regenerate this test case?

oharboe, 2023-07-21:
> @oharboe Can you post the full environment to regenerate this test case?

Follow instructions in link above.

osamahammad21, 2023-07-27:
@oharboe yeah but when I do it on master ORFS, I get the following error:

> [ERROR PDN-0179] Unable to repair all channels.

oharboe, 2023-07-27:
> @oharboe yeah but when I do it on master ORFS, I get the following error:
> 
> > [ERROR PDN-0179] Unable to repair all channels.

I see.... I can file a better github issue with commit hashes next time I see it... Otherwise you'll have to go spelunking in older versions of openroad and ORFS to reproduce the case....

closed 2023-07-27 by oharboe