The-OpenROAD-Project/OpenROAD #3649: Segfault during generate_abstract

oharboe, 2023-07-18:
### Describe the bug

```
Starting generation of abstract views
Signal 11 received
Stack trace:
 0# 0x0000000000D2A1A7 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FD5BD343400 in /lib64/libc.so.6
 2# odb::dbTechLayer::getType() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# odb::lefout::insertObstruction(odb::dbTechLayer*, odb::Rect const&, std::map<odb::dbTechLayer*, boost::polygon::polygon_90_set_data<int>, std::less<odb::dbTechLayer*>, std::allocator<std::pair<odb::dbTechLayer* const, boost::polygon::polygon_90_set_data<int> > > >&) const in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# odb::lefout::findInstsObstructions(std::map<odb::dbTechLayer*, boost::polygon::polygon_90_set_data<int>, std::less<odb::dbTechLayer*>, std::allocator<std::pair<odb::dbTechLayer* const, boost::polygon::polygon_90_set_data<int> > > >&, odb::dbBlock*) const in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# odb::lefout::getObstructions(odb::dbBlock*, std::map<odb::dbTechLayer*, boost::polygon::polygon_90_set_data<int>, std::less<odb::dbTechLayer*>, std::allocator<std::pair<odb::dbTechLayer* const, boost::polygon::polygon_90_set_data<int> > > >&) const in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# odb::lefout::writeObstructions(odb::dbBlock*) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# odb::lefout::writeBlock(odb::dbBlock*) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# odb::lefout::writeAbstractLef(odb::dbBlock*) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# odb::dbBlock::saveLef(char*, int, bool) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# 0x00000000011BF2A7 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# 0x0000000000D6CC1F in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
13# 0x00007FD5C205636C in /lib64/libtcl8.5.so
14# TclObjInterpProcCore in /lib64/libtcl8.5.so
15# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
16# 0x00007FD5C205636C in /lib64/libtcl8.5.so
17# 0x00007FD5C205E647 in /lib64/libtcl8.5.so
18# TclEvalObjEx in /lib64/libtcl8.5.so
19# 0x00007FD5C209927F in /lib64/libtcl8.5.so
20# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
21# 0x00007FD5C205636C in /lib64/libtcl8.5.so
22# 0x00007FD5C205E647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007FD5C20191D0 in /lib64/libtcl8.5.so
25# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
26# 0x00007FD5C205636C in /lib64/libtcl8.5.so
27# 0x00007FD5C205E647 in /lib64/libtcl8.5.so
28# TclEvalObjEx in /lib64/libtcl8.5.so
29# 0x00007FD5C201CF00 in /lib64/libtcl8.5.so
30# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
31# 0x00007FD5C205636C in /lib64/libtcl8.5.so
32# TclObjInterpProcCore in /lib64/libtcl8.5.so
33# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
34# 0x00007FD5C205636C in /lib64/libtcl8.5.so
35# TclObjInterpProcCore in /lib64/libtcl8.5.so
36# 0x00007FD5C2011EB2 in /lib64/libtcl8.5.so
37# 0x00007FD5C2012F1E in /lib64/libtcl8.5.so
38# Tcl_EvalEx in /lib64/libtcl8.5.so
39# Tcl_Eval in /lib64/libtcl8.5.so
40# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
41# ord::tclAppInit(Tcl_Interp*) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
42# Tcl_Main in /lib64/libtcl8.5.so
43# main in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
44# __libc_start_main in /lib64/libc.so.6
45# 0x0000000000D23EC7 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
```

### Expected Behavior

no crash

### Environment

```shell
latest master
```

### To Reproduce

Will work on whittling down an example, I can't share the failing design.

oharboe, 2023-07-18:
failed miserably with `deltaDebug.py`...

In debugger, I see:

[image]

maliberty, 2023-07-18:
Can you look at the LEF for the term of this master and see if there is a via there?  It seems odd to have a layer-less via.

PR #3657 (2023-07-18): odb: expand vias in iterms in lef out
Fixes #3649
commit 85f0adef (2023-07-18): odb: expand vias in iterms in lef out

Fixes #3649
closed 2023-07-18 by maliberty