The-OpenROAD-Project/OpenROAD #3653: Print executed TCL command back to stdout

Blebowski, 2023-07-18:
### Description

When running OR in "batch mode", with `<cmd_file>`, IMHO it would be good if OR printed each line
of the sourced TCL file before executing it. This-way it is much simpler to debug the script without
the need to manually print something before each executed command.

Or did I miss something in the manual and there is an option how to enable this type of behavior?

Is this feasible ? I guess implementation wise, its just a matter of one more "print" somewhere, but
I can imagine that it can break lot of tests if this behavior is made default.

Blebowski, 2023-07-18:
Ok, now I see that it is similar to:
https://github.com/The-OpenROAD-Project/OpenROAD/issues/3539

maliberty, 2023-07-18:
duplicate as noted

closed 2023-07-18 by maliberty