The-OpenROAD-Project/OpenROAD #3658: make write_lef and others write out files atomically

oharboe, 2023-07-18:
### Description

A half-written .lef file can be lying around after a crash which complicates e.g. `deltaDebug.py`.

Suggestion:

1. modify all `write_*` commands to write to a temp file
2. when the write completes, rename temp file to final name

This way there are no partial files lying around

annapetrosyan26, 2024-04-26:
Hello @maliberty @oharboe , I'd like to work on this issue. Could you please assign it to me ?

oharboe, 2024-04-26:
@annapetrosyan26 You can simply create a pull request without being assigned this task, I don't have the role to assign github issues in this project.

I haven't seen that @maliberty assigns github issues to new contributors, but perhaps that's something that would be approperiate?

annapetrosyan26, 2024-04-26:
Thank you @oharboe for the clarification. Then I'll work on the issue and once the code is ready I will create a pull request.

oharboe, 2024-07-14:
@maliberty Fixed, no?

maliberty, 2024-07-14:
Fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/5109

closed 2024-07-14 by maliberty