The-OpenROAD-Project/OpenROAD #3659: incremental global routing crashes when used with repair_timing -recover_power

openroadie, 2023-07-18:
### Describe the bug

```
Signal 11 received
Stack trace:
 0# 0x00005607F1141458 in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
 1# 0x00007FD639912090 in /lib/x86_64-linux-gnu/libc.so.6
 2# grt::IncrementalGRoute::updateRoutes() in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
 3# rsz::Resizer::updateParasitics() in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
 4# rsz::RecoverPower::recoverPower() in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
 5# 0x00005607F128312B in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
 6# TclNRRunCallbacks in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007FD63D5DD924 in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
10# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
11# ord::tclAppInit(Tcl_Interp*) in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
12# Tcl_MainEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
13# main in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad
14# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
15# _start in /home/harsh/work/harsh/or3/OpenROAD/build/src/openroad

```
As described in the title

### Expected Behavior

No crashes. It is possible that there are bugs in the recover_power code so please don't assume the netlist connectivity is correct (but in that case incremental global routing should error out instead of crashing). 

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.8.7-050807-lowlatency
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: da9d8fe190e8469f48662b401515870c371c0a3a
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /home/harsh/work/harsh/gcloud_test/.venv/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
[... 127 lines cut ...]
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk global_route

Actually any ORFS testcase will show the issue as long as the global_route.tcl file is modified as follows:

global_route -guide_file $env(RESULTS_DIR)/route.guide \
               -congestion_report_file $env(REPORTS_DIR)/congestion.rpt \
               {*}[expr {[info exists ::env(GLOBAL_ROUTE_ARGS)] ? $::env(GLOBAL_ROUTE_ARGS) : {-congestion_iterations 20 -verbose}
}]

set_propagated_clock [all_clocks]
estimate_parasitics -global_routing
**repair_timing -recover_power**

source $env(SCRIPTS_DIR)/report_metrics.tcl
report_metrics "global route"

### Relevant log output

```shell
N/A
```

### Screenshots

N/A

openroadie, 2023-07-18:
@maliberty @luis201420 @eder-matheus I do not know how to upload the issue tar file. I tried to drag it into this window but it got rejected since there is a limit of 25MB and my file is 27MB. You can run any design with the 1 line change in the "To Reproduce" section. asap7/gcd takes < 5 minutes to show the problem.

Let me know what I am missing in terms of being to upload a tar.gz file with the testcase. I used the latest OR and ORFS

maliberty, 2023-07-18:
You can put it in gdrive and put a link here.

openroadie, 2023-07-19:
Thank you @maliberty . I should have thought of that. Here it is:

https://drive.google.com/file/d/1kOHD0lI8kluOMyX_hzB6KisN7ULFTI__/view?usp=sharing

eder-matheus, 2023-07-31:
> @luis201420 now I have a different crash in antenna rule fixing. Here is the testcase:
> 
> https://drive.google.com/file/d/1GKiQ9ee0sd8KVr74PeM0MYhkzDCBu7qX/view?usp=sharing

@openroadie Could you create a new issue for this antenna crash? Please feel free to assign it to me. Also, can this issue be closed if the incremental routing crash is fixed?

openroadie, 2023-07-31:
Verified that the original problem is fixed. Thank you @luis201420

closed 2023-07-31 by openroadie