The-OpenROAD-Project/OpenROAD #3661: add_corners allowed multiple times

Blebowski, 2023-07-19:
### Describe the bug

When I try `define_corners`, it allows the same corners to be defined multiple times, e.g.:

```
define_corners min typ max
define_corners min typ max
define_corners min typ max
```

### Expected Behavior

I would expect that if I attempt to redefine already existing corner, that I get an error saying that the corner
already exists.

### Environment

```shell
Git commit: 914d6d23c442175889a2cca9b8e6fc37fe66e244
kernel: Linux 4.18.0-425.10.1.el8_7.x86_64
os: Red Hat Enterprise Linux 8.7 (Ootpa)
cmake version 3.23.2
-- The CXX compiler identification is GNU 8.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9373-g8c6f0f8f9
-- System name: Linux
-- Compiler: GNU 8.5.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /tools/tropic/ts-python-env/virtualenvs/ts-python-env-yQ9mF3ZL-py3.8/bin/python3.8 (found version "3.8.13") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 93 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.MiJJvRuW5V
```

### To Reproduce

./openroad -gui

In TCL shell:

```
define_corners min typ max
define_corners min typ max
```

openroadie, 2023-07-21:
@Blebowski The current behavior is to clear any corner definitions from the previous define_corners command. This is what the STA documentation states:

"The define_corners command must follow set_operating_conditions -analysis_type and precede any reference to the corner names and can only appear once in a command file. There is no support for re-defining corners."

So the same corner is not being defined multiple times and only one define_corners command is allowed. In general it is better for tools to not error out so the decision here is to apply the last corners command. This could come in handy when there is a complex set of scripts and a lower level script may want to define corners different from what has already been defined. 

Not sure if anything needs to be done. If it is really critical ... maybe a warning can be added that the command that is meant to be used once is being used multiple times. Let me know what you think.

Blebowski, 2023-07-21:
Hi @openroadie , 

I see, so next calls "replace" all corners from previous calls. I think throwing a warning that states something like:
```
WARNING: "define_corners" executed when corners have previously been defined. Removing previous corners <LIST_OF_ORIGINAL_CORNERS> and replacing them with <LIST_OF_NEW_CORNERS>.
```
would be usefull here.

Also, I was wondering whether there could be a command `report_corners`. It could list the existing corners, along
with libraries associated with each corner and some information (e.g. operating conditions).

openroadie, 2023-07-21:
To be absolutely precise... a "define_corners" command will erase/clear all corners from the previous call... which erased all corners from the call before that and son on. 

Let me look at the feasibility of the warning as well as the report_corners command.

jjcherry56, 2023-07-21:
It isn't an error to redefine existing corners and there is no point in warning that the corners are being redefined. That is like saying "set x "a b c"; set x "e f g" should warn that x is being redefined.
If you try to call it after a reading liberty file it will error because there is no support for
redefining corners.

Blebowski, 2023-07-21:
Hi @jjcherry56 , OK then.

However, if the `define_corners` should be called only once, is following valid:
```
define_corners max
define_corners min
```
or further (since you said error will be thrown if called after liberty was loaded), following:
```
define_corners max
read_liberty -corner max <lib_file>
define_corners min
read_liberty -corner min <lib_file>
```
?

jjcherry56, 2023-07-21:
no, it is not valid.
why don't you try running it.

Blebowski, 2023-07-21:
Thanks. I am not on the PC now where I have those libs available. I would do that otherwise.

jjcherry56, 2023-07-21:
The libraries are irrelevant. You can use the ones referenced in examples/multi_corner.tcl

Blebowski, 2023-07-21:
Yup, you are right, I should have checked.

closed 2023-12-10 by Blebowski