The-OpenROAD-Project/OpenROAD #3674: cts inserting buffers on specials nets and RDL routing to pads

gadfort, 2023-07-20:
### Describe the bug

When the clock pin is defined as a wirebond or bump cell input for a chip, CTS will insert a clock buffer on the pad itself which causes the design to unrouteable and connecting cells directly to IO voltages.

### Expected Behavior

CTS does not modify pad connections

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.26.4
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9331-g536692015
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 79 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (31.1s)
-- Generating done (0.6s)
```

### To Reproduce

[sc_issue_zerosoc_signal_m4_cts0_20230720-081826.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12109424/sc_issue_zerosoc_signal_m4_cts0_20230720-081826.tar.gz)

```
tar xvf sc_issue_zerosoc_signal_m4_cts0_20230720-081826.tar.gz
cd sc_issue_zerosoc_signal_m4_cts0_20230720-081826
./run.sh
```

inspect port "we_pad\[5\]", this is no longer connected to the padring.

maliberty, 2023-07-20:
Do you have a .lib for the pad cell?  I'm not finding it in your test case

gadfort, 2023-07-20:
@maliberty it's included as " ../../sc_collected_files/sky130_dummy_io_fd3d9afce7ba0ddbf6fb0d6459e4930dc00a120b.lib" I just checked and reran the test and it's there.

gadfort, 2023-07-20:
@maliberty thats interesting. I see what you mean, I'll have to look into this, but the underlying issue still stands (I has this issue too in last years tapeout).

maliberty, 2023-07-20:
Agreed but just FYI so that when this is fixed you don't wonder why you still have no clock tree.

gadfort, 2023-07-20:
@maliberty yes, that would also be very confusing (thanks), I probably never noticed because to get around the CTS thing, the clock was always defined to the output pin of the pad.

PR #3678 (2023-07-20): cts: skip special nets (eg RDL routing from bump to pad)
[INFO CTS-0116] Special net "we_pad[5]" skipped.

Fixes #3674
commit 06c5991a (2023-07-20): cts: skip special nets (eg RDL routing from bump to pad)

[INFO CTS-0116] Special net "we_pad[5]" skipped.

Fixes #3674
closed 2023-07-20 by maliberty