The-OpenROAD-Project/OpenROAD #3680: Default to monospace font in OpenROAD GUI

oharboe, 2023-07-21:
### Description

This will make it easier to read e.g. `report_checks`.

[image]

oharboe, 2023-07-21:
It doesnt work in Op Docker GUI

oharboe, 2023-07-21:
It works locally, but not when run from within Docker

maliberty, 2023-07-21:
Perhaps you have no monospace font installed in your container

oharboe, 2023-07-21:
> Perhaps you have no monospace font installed in your container

I build the container with `./build_openroad.sh`

maliberty, 2024-12-26:
@jeffng-or since you are working with docker perhaps you can look at this.  I suspect some font just needs to be installed in the container.