The-OpenROAD-Project/OpenROAD #3692: build fails when running cmake .

proppy, 2023-07-21:
### Describe the bug

build fails when running cmake from the main source directory with the following error:
```
[ 74%] Linking CXX static library ifp.a
Error: IFP 0038 used 2 times
cmamake[2]: *** [src/ifp/src/CMakeFiles/ifp.dir/build.make:157: src/ifp/src/ifp.a] Error 1
```

### Expected Behavior

build passes

### Environment

```shell
N/A
```

### To Reproduce

```
cmake .
make
```

### Relevant log output

```shell
This is likely due to `find_message.py` inspecting generated files for similar error messages, see:

IFP 0038 InitFloorplan.i:62        No design is loaded. ERROR https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/home/proppy/src/github.com/The-OpenROAD-Project/OpenROAD/src/ifp/src/InitFloorplan.i#L62
IFP 0038 InitFloorplanTCL_wrap.cxx:1792 No design is loaded. ERROR https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/home/proppy/src/github.com/The-OpenROAD-Project/OpenROAD/src/ifp/src/CMakeFiles/ifp.dir/InitFloorplanTCL_wrap.cxx#L1792
```
```

vvbandeira, 2023-07-22:
@proppy I only see the message in one place:
```
src/ifp/src/InitFloorplan.i:    logger->error(utl::IFP, 38, "No design is loaded.");
```
Could you try a clean build? Might be a caching issue.

maliberty, 2023-10-17:
stale

closed 2023-10-17 by maliberty