The-OpenROAD-Project/OpenROAD #3719: Support routing by abutment in mpl2

oharboe, 2023-07-25:
### Description

mock-array uses manual placement to achieve horizontal routing by abutment:

[image]

To see if MPL2 can do routing by abutment, make the following changes:

```
diff --git a/flow/designs/asap7/mock-array/config.mk b/flow/designs/asap7/mock-array/config.mk
index cf124242..6e1b4abf 100644
--- a/flow/designs/asap7/mock-array/config.mk
+++ b/flow/designs/asap7/mock-array/config.mk
@@ -28,7 +28,9 @@ BLOCKS                       = Element
 
 export GDS_ALLOW_EMPTY       = Element
 
-export MACRO_PLACEMENT_TCL   = ./designs/asap7/mock-array/macro-placement.tcl
+# export MACRO_PLACEMENT_TCL   = ./designs/asap7/mock-array/macro-placement.tcl
+export SYNTH_HIERARCHICAL = 1
+export RTLMP_FLOW = True
 
 export IO_CONSTRAINTS        = designs/asap7/mock-array/io.tcl
 
```

Then run `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk`:

Macro placement fails:

```
The design only has macros.

[Hier-RTLMP::HardMacroClusterMacroPlacement] Place macros in cluster: root
[ERROR MPL-0010] Cannot find valid macro placement for hard macro cluster: root
Error: macro_place.tcl, 141 MPL-0010
Command exited with non-zero status 1
Elapsed time: 0:07.66[h:]min:sec. CPU time: user 59.87 sys 0.03 (781%). Peak memory: 234256KB.
make[1]: *** [Makefile:523: do-2_4_floorplan_macro] Error 1
```

### Suggested Solution

It would be nice if MPL2 could do routing by abutment. It is very tricky to get manual placement exactly right, lots of alignment requirements...

maliberty, 2025-08-30:
@oharboe still relevant?

oharboe, 2025-08-30:
mpl2 has been doing routing by abutment for a year or so?

closed 2025-08-30 by oharboe