The-OpenROAD-Project/OpenROAD #3724: add_pdn_stripe incorrectly checks if a stripe fits into boundary of the selected grid

chaufe, 2023-07-26:
### Describe the bug

`add_pdn_stripe` checks if a set of stripes can be added into the boundaries of a grid.
Due to a math error in this check, no stripe set can be added exactly at the right or top boundary of a grid, but needs to have spacing of half of the stripe width between the last stripe and the right or top boundary.

On the other hand, `add_pdn_stripe` allows stripes to be created up to the half of the width of a stripe outside of the left or bottom boundary of a grid.

I assume, `add_pdn_stripe` performs its range checks based on the left or bottom side of a stripe to be created rather than based on the center of the first stripe of a set of stripes. This shifts the comparison point by half of the stripe's width.

This error occurs for both stdcell grid and macro grid.

I've attached a test case where a vertical stripe gets incorrectly created outside the grid boundary, and where a vertical pair of stripes ending exactly on the right boundary of the grid does not get created with the following error:
`[ERROR PDN-0185] Insufficient width (189.28 um) to add straps on layer Metal4 in grid "stdcell_grid" with total strap width 8.53 um and offset 182.75 um.`
The grid has a width of 189.28um, the two stripes have a width of 4um each and a spacing of 0.530um, so given `-offset` refers to the center of the left stripe, a spacing of 6.53um to the right boundary should suffice to fit both stripes into the grid area, not 8.53um as indicated by the error message.

I have additionally attached an ipython notebook for Google Colab to reproduce the issue.

/cc @proppy
/cc @dhaentz1

### Expected Behavior

`add_pdn_grid` should take into consideration, that its option `-offset` refers to the center of the first stripe to be created, but not to the left/bottom side.

Given a die area and core area of `[0 0 189.28 188.16]`, the following commands should fail:
```
# creating vertical stripe pair, first stripe with center x=0, 
#   left edge starts 2um outside grid boundary (x=-2um)
# -> currently, this does not fail
add_pdn_stripe \
        -grid stdcell_grid \
        -layer $::env(FP_PDN_LOWER_LAYER) \
        -width 4.0 \
        -offset 0.0 \
        -pitch 50 \
        -spacing 0.530 \
        -number_of_straps 1 \
        -starts_with POWER

# creating horizontal stripe pair, first stripe with center y=0, 
#    bottom edge starts 2um outside grid boundary (y=-2um)
# -> currently, this does not fail
add_pdn_stripe \
        -grid stdcell_grid \
        -layer $::env(FP_PDN_UPPER_LAYER) \
        -width 4.0 \
        -offset 0.0 \
        -pitch 50 \
        -spacing 0.530 \
        -number_of_straps 1 \
        -starts_with POWER
```

Given a die area and core area of `[0 0 189.28 188.16]`, the following commands should **not** fail:
```
# creating vertical stripe pair, first stripe with center x=182.75um, 
#    right edge of last stripe ends at right boundary of grid
# -> currently, this does fail
add_pdn_stripe \
        -grid stdcell_grid \
        -layer $::env(FP_PDN_LOWER_LAYER) \
        -width 4.0 \
        -offset [expr 189.28 - 6.53]\
        -pitch 50 \
        -spacing 0.530 \
        -number_of_straps 1 \
        -starts_with POWER

# creating horizontal stripe pair, first stripe with center y=182.75um, 
#    right edge of last stripe ends at top boundary of grid
# -> currently, this does fail
add_pdn_stripe \
        -grid stdcell_grid \
        -layer $::env(FP_PDN_UPPER_LAYER) \
        -width 4.0 \
        -offset [expr 188.16 - 6.53]\
        -pitch 50 \
        -spacing 0.530 \
        -number_of_straps 1 \
        -starts_with POWER

```

### OpenROAD Environment

```shell
Env.sh seems not to be included in my litex-hub package - please use the testcase notebook to extract data.
```

### OpenLane Environment

```shell
env.py seems to be broken in my setup - please use the testcase notebook to extract data.
```

### To Reproduce

[OpenRoad__math_bug_in_pdn.ipynb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12174075/OpenRoad__math_bug_in_pdn.ipynb.gz)

[issue_reproducible.pdn_math_error.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12174093/issue_reproducible.pdn_math_error.zip)

### Relevant log output

```shell
[ERROR]: Last 10 lines:
Creating horizontal stripe pair starting 2um left of core area.
This should fail as offset is 0 and width is 4.0um, so strap starts at x == -2.0um
... but it does not fail
Creating horizontal strip pair ending exactly at right end of core area.
This should not fail as offset 182.75um + 6.53um == 189.28um (width of core area)
6.53um = (1/2 + 1) * stripe_width + stripe-to-stripe spacing
... but it does fail
[ERROR PDN-0185] Insufficient width (189.28 um) to add straps on layer Metal4 in grid "stdcell_grid" with total strap width 8.53 um and offset 182.75 um.
Error: custom_pdn_cfg.tcl, 42 PDN-0185
child process exited abnormally
```

gadfort, 2023-07-26:
@chaufe this testcase is not runnable. Please add the missing variables to and reupload.

chaufe, 2023-07-27:
@gadfort I'm using the "stock" issue_reproducible created from within OpenLane's `or_issue.py`, but it seems only a subset of the variables of my `config.tcl` have been included, i.e. `FP_PDN_LOWER_LAYER` has been included into `run.sh` but others like `FP_PDN_UPPER_LAYER` are missing. Flow and PDK defaults are also missing. Re-creating the issue_reproducible - even using the latest version of `or_issue.py` from github - renders the same result. 

Would my `config.tcl` be of any help for you?

```
set ::env(DESIGN_NAME) "testcase"

set ::env(VERILOG_FILES) [list "testcase.v" ]

set ::env(DESIGN_IS_CORE) 1
set ::env(FP_PDN_CORE_RING) 0
set ::env(FP_PDN_UPPER_LAYER) Metal5
set ::env(FP_PDN_LOWER_LAYER) Metal4
set ::env(FP_PDN_RAILS_LAYER) Metal1

set ::env(CLOCK_TREE_SYNTH) 1
set ::env(CLOCK_PORT) "clk"
set ::env(CLOCK_PERIOD) 100

set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 189.28 188.16"
set ::env(CORE_AREA) "0 0 189.28 188.16"

set ::env(PL_TARGET_DENSITY) 0.65

# use custom PDN config
set ::env(PDN_CFG) /content/custom_pdn_cfg.tcl
```

Btw, my testcase can also be fully run inside Google Colab using this link:
https://colab.research.google.com/github/chaufe/testcases/blob/main/colab/OpenRoad__math_bug_in_pdn.ipynb

maliberty, 2023-07-27:
@donn please help with fixing or_issue.py so we can get a proper issue.

maliberty, 2023-10-17:
no usable test case provided

closed 2023-10-17 by maliberty