The-OpenROAD-Project/OpenROAD #3727: Don't write unrequested report in check_placement

maliberty, 2023-07-26:
### Describe the bug

In check_placement the code has:
```
  set file_name "failures_report.json"
  if { [info exists keys(-report_file_name) ] } {
    set file_name $keys(-report_file_name)
  }
```
This means failures_report.json is dumped in the working directory whenever no report is requested.  

### Expected Behavior

Please change this to not write anything if -report_file_name is not used.

### Environment

```shell
head of master
```

### To Reproduce

as above

maliberty, 2023-07-26:
@eder-matheus FYI

maliberty, 2023-07-26:
aside: please also make an ORFS PR to use -report_file_name

maliberty, 2023-08-01:
@mina1460 eta?

mina1460, 2023-08-01:
@maliberty 
today

closed 2023-08-01 by maliberty