The-OpenROAD-Project/OpenROAD #3737: Print warning when macros are not placed on grid

oharboe, 2023-07-28:
### Description

If you have a rounding error in placing the macros and are off by 1nm, you'll get strange DRC errors:

[image]

### Suggested Solution

Examine macro placements and print a warning if the macros are not "on the grid".

maliberty, 2023-07-28:
@osamahammad21 I thought you already did this in drt.

oharboe, 2023-07-28:
> @osamahammad21 I thought you already did this in drt.

Maybe it is there... there are a LOT of logs. There's no summary of warnings and a way to evaluate the warnings; Quartus/Intel has this in its FPGA tool...

Perhaps a beter user experience would be an error with a default and an option to ignore? :-)

osamahammad21, 2023-07-28:
It should print an error message according to this:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e07b05a950cc8eb5c2c088d6e10484f2aa8b1a5/src/drt/src/io/io.cpp#L152-L185
Could you share the test case or any way to reproduce it?

oharboe, 2023-07-28:
> It should print an error message according to this:
> 
> https://github.com/The-OpenROAD-Project/OpenROAD/blob/0e07b05a950cc8eb5c2c088d6e10484f2aa8b1a5/src/drt/src/io/io.cpp#L152-L185
> 
> 
> Could you share the test case or any way to reproduce it?

Those are errors: no I did not see them when the macros  were off by 1nm

oharboe, 2023-07-28:
@maliberty Hmmm.... deltaDebug.py cant check for the absence of an error...

oharboe, 2023-07-28:
@osamahammad21 Just modify mock-array to be off by 1nm?

maliberty, 2023-07-28:
asap7 manufacturing grid is 1nm so off but 1nm is not off grid.

oharboe, 2023-07-28:
> asap7 manufacturing grid is 1nm so off but 1nm is not off grid.

I meant that the pins, as you can see in the picture are off by 1nm. The macro needs to be aligned to 0.048nm or detailed routing ends in tears...

maliberty, 2023-07-29:
It may be a problem but it is not off grid.  We already have a check for off grid.

rovinski, 2023-07-29:
We have errors for if something is off the manufacturing grid. I think @oharboe is talking about if the pins are off the routing grid.
I think realistically, that's somewhat difficult to check in all cases but might be possible if the pins are exactly min width.

maliberty, 2023-07-29:
@rovinski I think that is a reasonable warning if we have a macro with min-width single-rectangle pins that are not on the routing grid.

osamahammad21, 2023-07-29:
@maliberty but theoretically speaking, aren't off-routing-grid pins routable? They are hard to route by the current implementation, but in the end you could get a clean routing solution.

maliberty, 2023-07-29:
@osamahammad21 yes but since we do poorly a warning to that effect is useful.

osamahammad21, 2023-07-31:
@maliberty Ok. Should I fire a warning after pin access if there is a pin with empty access pattern or should it be earlier in io?

maliberty, 2023-07-31:
io since it doesn't depend on pa

closed 2023-08-15 by maliberty