The-OpenROAD-Project/OpenROAD #3739: Right click on timing path and inject OpenSTA command to report that path

oharboe, 2023-07-28:
### Description

Today I have to manually type up the report_checks command for a timing path if I want to drill down for more information or create some variant.

### Suggested Solution

It would be nice to be able to right click on a path and have an OpenSTA command injected into the console that the user can then refine.

AcKoucher, 2023-08-03:
@oharboe So the idea is that, when right clicking a selected path in Timing Report, the command should be "written" in the scripting widget's TCL commands space for the user to refine? Or the report_checks which corresponds to the right clicked path should run directly in the console?

oharboe, 2023-08-04:
> @oharboe So the idea is that, when right clicking a selected path in Timing Report, the command should be "written" in the scripting widget's TCL commands space for the user to refine?

Yes.

AcKoucher, 2023-08-07:
Something like:
`report_checks -from <start> -to <end>`

would be enough?

oharboe, 2023-08-07:
> Something like: `report_checks -from <start> -to <end>`
> 
> would be enough?

yes. though i would add all the fields and options so the user can just delete what he doesnt need or want.

This is "marketing" of the underlying fearure as well as convenience.

maliberty, 2023-08-07:
You'll want -path_delay (min|max) depending on the tab.  Also "-fields {slew cap input nets fanout} -format full_clock_expanded" are closer to what the GUI shows.

PR #3818 (2023-08-14): gui: add context menu in Timing Report for "writing" report_checks in tcl input
Resolves #3739
closed 2023-08-14 by maliberty