The-OpenROAD-Project/OpenROAD #3752: ZASSERT error in check_antenna

openroadie, 2023-08-01:
### Describe the bug

Running the debug build, OR asserts in antenna checking in this code:

void dbWireEncoder::newPathVirtualWire(int jct_id,
                                       dbTechLayer* layer,
                                       dbWireType type)
{
  **ZASSERT((jct_id >= 0) && (jct_id < _idx));**

### Expected Behavior

No assertion error. The latest master should show this issue. 

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.8.7-050807-lowlatency
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:98 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: ab7625fe5843c6579d28068b2a7e4dfb38f9dcb5
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 99 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.RuluGbfNmq
```

### To Reproduce

Run the provided testcase that is available here:

https://drive.google.com/file/d/1GKiQ9ee0sd8KVr74PeM0MYhkzDCBu7qX/view?usp=sharing

### Relevant log output

```shell
[WARNING ANT-0011] -report_violating_nets is deprecated.
[INFO GRT-0043] No OR_DEFAULT vias defined.
openroad: /home/harsh/work/harsh/or3/OpenROAD/src/odb/src/db/dbWireCodec.cpp:481: void odb::dbWireEncoder::newPathVirtualWire(int, odb::dbTechLayer*, odb::dbWireType): Assertion `(jct_id >= 0) && (jct_id < _idx)' failed.
Signal: SIGABRT (hit program assert)
```

openroadie, 2023-08-01:
@eder-matheus Even if I comment out the "repair_timing -recover_power" command in global_route.tcl ... the problem still occurs. Just FYI. So this does not seem to be related to power recovery operations (downsize).

eder-matheus, 2023-08-01:
PR created. I've tested with the provided test case, and the assertion failure no longer happens.

openroadie, 2023-08-01:
Thank you for the lightening fast turnaround :)

closed 2023-08-01 by maliberty