The-OpenROAD-Project/OpenROAD #3753: DRT "Non orthogonal wire" strikes again

smunaut, 2023-08-01:
### Describe the bug

Similar to #3385 but in different cases because here there are no vias.
But it still hallucinates non existent "Non orthogonal wire".

So as a reminder, I do some custom manual routing using the ODB API before calling the GRT/DRT to perform routing of the rest of the nets. And somehow DRT doesn't like some of the routing I did.

This is the reported error : 
```
[ERROR DRT-1010] Unsupported non-orthogonal wire begin=(889.18, 159.62) end=(892.86, 160.98), layer met4
```

And in the DEF after my custom routing this is the relevant segment routing.

```
      + FIXED met4 ( 863420 159620 ) ( * 160980 )
      NEW met4 ( 867100 159620 ) ( * 160980 )
      NEW met4 ( 870780 159620 ) ( * 160980 )
      NEW met4 ( 874460 159620 ) ( * 160980 )
      NEW met4 ( 878140 159620 ) ( * 160980 )
      NEW met4 ( 881820 159620 ) ( * 160980 )
      NEW met4 ( 885500 159620 ) ( * 160980 )
      NEW met4 ( 889180 159620 ) ( * 160980 )
      NEW met4 ( 892860 159620 ) ( * 160980 )
      NEW met4 ( 896540 159620 ) ( * 160980 )
      NEW met4 ( 900220 159620 ) ( * 160980 )
      NEW met4 ( 903900 159620 ) ( * 160980 )
      NEW met4 ( 907580 159620 ) ( * 160980 )
      NEW met4 ( 911260 159620 ) ( * 160980 )
      NEW met4 ( 914940 159620 ) ( * 160980 )
      NEW met4 ( 918620 159620 ) ( * 160980 )
      NEW met4 ( 922300 159620 ) ( * 160980 )
      NEW met4 ( 925980 159620 ) ( * 160980 )
      NEW met4 ( 929660 159620 ) ( * 160980 )
      NEW met4 ( 933340 159620 ) ( * 160980 )
      NEW met4 ( 937020 159620 ) ( * 160980 )
      NEW met4 ( 940700 159620 ) ( * 160980 )
      NEW met4 ( 944380 159620 ) ( * 160980 )
      NEW met4 ( 948060 159620 ) ( * 160980 )
      NEW met4 ( 859740 159620 ) ( * 160980 )
      NEW met4 ( 859740 159620 ) ( 948060 * ) ;
```

But that segment it complains about doesn't exist ...

Also, if I remove the last segment in the above routing (the horizontal one), then it works just fine ... no errors.

Reproducer :

https://people.osmocom.org/tnt/stuff/tt/tcl_reproducible.zip

### Expected Behavior

Well, I expect no errors and for DRT to complete.

### Environment

```shell
Git commit: 797b89e4e26b29f0bd89a58eff1f8258de08c7d3
kernel: Linux 5.10.0-19-cloud-amd64
os: Debian GNU/Linux 11 (bullseye)

... it doesn't complete because that script expects to find some package which are not in the default path, I need to give argument to cmake for it to find them ... but obviously build completed fine so it's the Env.sh script which has an issue ...

% tnt@vm-tt:/mnt/pdk/OL2/native/OpenROAD $ ./_build/src/openroad  -version
v2.0-9635-g797b89e4e 
```
```

### To Reproduce

See the reproduced posted above. Can't attach it here, it's too big for github.

smunaut, 2023-08-01:
I'm reading `io::Parser::setNets` in  `src/drt/src/io/io.cpp` and I can't figure out how this is even supposed to work.

This is an excerpt of a path in the DEF :   (DRT actually reads it from ODB but DEF is readable by human so not bad to see what's in there).

```
      + FIXED met4 ( 8740 214380 ) ( * 215740 ) ( 12420 * ) ( * 214380 )
      NEW met4 ( 12420 215740 ) ( 16100 * ) ( * 214380 )
      NEW met4 ( 16100 215740 ) ( 19780 * ) ( * 214380 )
     ....
```

Now here is some debug output I generated in that function. I just print the `pathId` textual representation for each element parsed from the wireDecoder and if it's a POINT I also print out the X/Y coordinate for it.
Then I also print a "SEGMENT" for every `frPathSeg` created and added to the net.

```
OP: PATH
OP: POINT 8740 214380 (begin)
OP: ITERM
OP: POINT 8740 215740 (end)
OP: POINT 12420 215740 (end)
OP: POINT 12420 214380 (end)
OP: ITERM
SEGMENT ( 8740 214380 ) ( 12420 214380 )
OP: JUNCTION
OP: POINT 12420 215740 (begin)
OP: POINT 16100 215740 (end)
OP: POINT 16100 214380 (end)
OP: ITERM
SEGMENT ( 12420 215740 ) ( 16100 214380 )
```

So we can see that the `OP` lines do match the DEF pretty well so the wiredecoder provides good output and all the point describe orthogonal segments.

HOWEVER if we look at the actual `SEGMENT` created ... they don't match at all what's read from the wire decoder ...
So yeah you end up with diagonal stuff. But even when it's not diagonal, it doesn't even always match the actual real segments.

I think this also explain a periodic problem I have with DRT routing stuff over my manually routed wire. Because it's internal view of the routed segments doesn't actually match what's in the database whatsoever.

eder-matheus, 2023-08-02:
@smunaut I've created a PR with the fix. I have tested it with the provided test case and now it's working.

The problem was that DRT was not detecting the non-orthogonal connections you have in met4 as different segments. It was "skipping" the points in the middle, leading to the non-orthogonal error. Now DRT is checking when we have a direction change in a sequence of points in the same layer, creating different segments for each part of the connection.

smunaut, 2023-08-03:
Well, although it's not crashing / error-ing out, I don't think it does what it should.

This is what was happening before :
```
OP: PATH
OP: POINT 859740 160980 (begin)
OP: ITERM
OP: POINT 859740 159620 (end)
OP: POINT 863420 159620 (end)
OP: POINT 863420 160980 (end)
OP: ITERM
SEGMENT ( 859740 160980 ) ( 863420 160980 )
OP: JUNCTION
...
```

And indeed it's only creating the PathSeg basically from the start point to the end point directy, discarding all the intermediate ones.

And this is what it does now :   (notice the "input" from the wiredecoder, the OP lines, are the same ).

```
OP: PATH
OP: POINT 859740 160980 (begin)
OP: ITERM
OP: POINT 859740 159620 (end)
SEGMENT ( 859740 160980 ) ( 859740 159620 )
OP: POINT 863420 159620 (begin)
OP: POINT 863420 160980 (end)
OP: ITERM
SEGMENT ( 863420 159620 ) ( 863420 160980 )
OP: JUNCTION
```

And so, it created 2 segments but it didn't create the segment from `( 859740 159620 )` to `( 863420 159620 )`.
So although it's making only orthogonal segments, it's still skipping some ...

smunaut, 2023-08-03:
```diff
diff --git a/src/drt/src/io/io.cpp b/src/drt/src/io/io.cpp
index 0e86dc1a7..b4ab3d990 100644
--- a/src/drt/src/io/io.cpp
+++ b/src/drt/src/io/io.cpp
@@ -674,7 +674,10 @@ void io::Parser::setNets(odb::dbBlock* block)
         top = -1;
         width = 0;
         bool endpath = false;
+
         do {
+          const char *ops[] = { "PATH", "JUNCTION", "SHORT", "VWIRE", "POINT", "POINT_EXT", "VIA", "TECH_VIA", "RECT", "ITERM", "BTERM", "RULE", "END_DECODE" };
+             printf("OP: %s%s", ops[pathId], pathId != odb::dbWireDecoder::POINT ? "\n":" ");
           switch (pathId) {
             case odb::dbWireDecoder::PATH:
             case odb::dbWireDecoder::JUNCTION:
@@ -690,9 +693,11 @@ void io::Parser::setNets(odb::dbBlock* block)
               if (!hasBeginPoint) {
                 decoder.getPoint(beginX, beginY);
                 hasBeginPoint = true;
+                printf("%d %d (begin)\n", beginX, beginY);
               } else {
                 decoder.getPoint(endX, endY);
                 hasEndPoint = true;
+                printf("%d %d (end)\n", endX, endY);
               }
               break;
             case odb::dbWireDecoder::POINT_EXT:
@@ -774,6 +779,7 @@ void io::Parser::setNets(odb::dbBlock* block)
           continue;
         }
         if (hasEndPoint) {
+          printf("SEGMENT ( %d %d ) ( %d %d )\n", beginX, beginY, endX, endY);
           auto tmpP = make_unique<frPathSeg>();
           if (beginX > endX || beginY > endY) {
             tmpP->setPoints(Point(endX, endY), Point(beginX, beginY));
```

eder-matheus, 2023-08-03:
@smunaut Thanks for the reply. I didn't notice this bug before. My last commit should fix this bug. I've printed the points and the created segments, and now all segments are correctly created.

smunaut, 2023-08-03:
@eder-matheus Yup it looks like it's doing the right thing now :+1:

closed 2023-08-07 by maliberty