The-OpenROAD-Project/OpenROAD #3754: [ERROR DRT-0085] Valid access pattern combination not found for <instance_name>

techstuds, 2023-08-01:
### Describe the bug

I'm facing this error at the pin_access step in global routing. The corresponding cells do not overlap with others and have enough space around them for routing. There are many other instances with the same master where everything went well. The design has 15192 unique inst patterns (see routing.log) and for some reason the router is having trouble accessing the pins of a few cells. If I run it with a netlist that doesn't have those cells in it, the error appears on another one.

### Expected Behavior

Complete pin_access and go forward with global routing.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.88.1.el7.x86_64
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9405-g9b378ff51
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 910 lines cut ...]
[INFO DRT-0082]   Complete 2000 groups.
[INFO DRT-0082]   Complete 3000 groups.
[INFO DRT-0082]   Complete 4000 groups.
[INFO DRT-0082]   Complete 5000 groups.
[INFO DRT-0082]   Complete 6000 groups.
[INFO DRT-0082]   Complete 7000 groups.
[INFO DRT-0082]   Complete 8000 groups.
[INFO DRT-0082]   Complete 9000 groups.
[INFO DRT-0083]   Complete 10000 groups.
@@@ dead end inst
@@@ dead end inst
[ERROR DRT-0085] Valid access pattern combination not found for 
pipe_g54590__5477
@@@ dead end inst
@@@ dead end inst
[ERROR DRT-0085] Valid access pattern combination not found for 
pipe_g54134__2883
[INFO DRT-0083]   Complete 20000 groups.
[INFO DRT-0083]   Complete 30000 groups.
Error: flow.tcl, 197 DRT-0085
```

### Additional Context

Has anyone encountered this error before and know how to deal with it?

vijayank88, 2023-08-01:
@techstuds 
Which PDK it is?

techstuds, 2023-08-01:
It's the XH018 PDK of X-FAB.

QuantamHD, 2023-08-01:
OpenROAD is having a hard time accessing these cell.

[WARNING DRT-0087] No valid pattern for unique instance pipe_g54590__5477, master is **AND4JIHDX0**
[WARNING DRT-0087] No valid pattern for unique instance pipe_g54134__2883, master is **ON31JIHDX1**

The easiest thing to do would be to "not use" those cells. You can mark them dont_use in your liberty.

gadfort, 2023-08-01:
I have a similar issue, delating to the number of vias the PA attempts. Here is my attempt to fix: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3677 but this seems to have impacts on the routablity of the design so more work is needed. With this, I can get my designs to route in my PDK.

maliberty, 2023-08-01:
What flow are you using?  In ORFS you can use 'make global_route_issue' to package up your test case (assuming it is ok to share).

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/deltaDebug.py is useful for cutting down a large test case if needed.

osamahammad21, 2023-08-14:
@techstuds Is it ok to share the test case?

techstuds, 2023-08-21:
@QuantamHD I've tried several times but the error still occurs until dont_use is ignored during synthesis (cells were marked as dont_use in SDC file before synthesis).

@gadfort Unfortunately, this didn't fix the error in my case, although it's a really promising change.

@maliberty I used the OpenROAD flow (not ORFS) in OpenROAD/test/flow.tcl because I'm mostly interested in P&R (the synthesis was done by Cadence Genus). 

I've also tried to cutting down my testcase with deltaDebug.py but I'm probably doing something wrong here. In flow.tcl, a db file is written just before the pin_access step where the error occurs, so I used that db file. What do I need to pass as a step argument in this case? The command looks like this:

`openroad -python deltaDebug.py --base_db_path test_dpl-tcl.db --error_string '[WARNING DRT-0087] No valid pattern for unique instance' --step step.sh --persistence 5`

step.sh:
`pin_access -bottom_routing_layer MET2 -top_routing_layer METTP`

osamahammad21, 2023-11-23:
@techstuds recently we added support for extensive via search for access points that can't be accessed with the conventional openroad pin access technique. It solved the issue for peter. Could you try it again and tell us what you get.
Also, if it is not solved yet, I think we need more information to work on this. At least some visuals on the pin that can't be accessed in relation to the routing tracks.

maliberty, 2024-03-02:
Closing due to no response.  If this is still relevant please reopen and update.

closed 2024-03-02 by maliberty