The-OpenROAD-Project/OpenROAD #3768: Add command to highlight only the IO nets

eder-matheus, 2023-08-02:
### Description

When debugging results from the pin placer, it is helpful to see the fly wires of the IO nets after the pin placement. I have to manually select all the IO pins to see these fly wires.

Since it would be used mainly during developer debugging, a command to highlight only the IO nets should be enough to simplify this process.

### Suggested Solution

This command can live inside PPL using the existing IOPlacerRendered. I suggest having something like `highlight_io_nets <net_names>`. This way, we can draw the fly wires for a subset of nets and not only for all nets. The `*` symbol can be used to draw the fly wires for all nets.

Some examples:
`highlight_io_nets {net1}` - Draw the fly wires of net1
`highlight_io_nets {net1 net2 net3}` - Draw the fly wires of the nets net1, net2 and net3
`highlight_io_nets {net*}` - Draw the fly wires of all nets with the prefix `net`.
`highlight_io_nets {*}` - Draw the fly wires of all nets

maliberty, 2023-08-02:
"Draw the fly wires of all nets" - should this be all IO nets?

maliberty, 2023-08-02:
Would it suffice to simply have a command to select IO nets without writing a new renderer?

eder-matheus, 2023-08-02:
> "Draw the fly wires of all nets" - should this be all IO nets?

Yes, I didn't notice I missed the IO part. Updated the description.

eder-matheus, 2023-08-02:
> Would it suffice to simply have a command to select IO nets without writing a new renderer?

Yeah, I think it should work. I've suggested using the IOPlacerRenderer, but I'm fine if we have a simpler solution for this.

maliberty, 2023-08-02:
The gui already has a "select" command.  Perhaps we can just add a filter option for IO nets.  

@gadfort any thoughts on generic filtering options?

maliberty, 2023-08-03:
perhaps @AcKoucher could add -filter attribute==value

AcKoucher, 2023-08-04:
> -filter attribute==value

value being a string?

maliberty, 2023-08-05:
value's type would depend on the attribute (ie -filter fanout=5 or -filter type=fill)

PR #3821 (2023-08-18): gui: add new attribute filtering for selection
Create a GUI selection filter based on the properties of the selected object.

The filter covers also the cases in which the property holds a SelectionSet or a PropertyList.

I created a special case to resolve #3768.
closed 2023-08-18 by maliberty