The-OpenROAD-Project/OpenROAD #3781: `odb`:: Error when the `dbNet` is connected with an `dbInstance` in the another `dbBlock`

ApeachM, 2023-08-03:
### Describe the bug

Hi. I want to report an error with `odb`. If a net is connected with a `dbIterm` which is in another block, then something happens in an unexpected way.

This is the test code.

You can reproduce this error with this [commit](https://github.com/ApeachM/OpenROAD/commit/79a0c542fd4d1facbd65ea0bb717a71d54196d0c).

```cpp
#include <iostream>

#include "db.h"
#include "defin.h"
#include "lefin.h"
#include "lefout.h"

using namespace odb;  // NOLINT(*-build-using-namespace)
using namespace std;  // NOLINT(*-build-using-namespace)
int main()
{
  dbDatabase* db_database = dbDatabase::create();
  dbTech* tech1 = dbTech::create(db_database, "top_tech");
  dbTech* tech2 = dbTech::create(db_database, "bottom_tech");
  auto* layer1
      = dbTechLayer::create(tech1, "top_layer", dbTechLayerType::MASTERSLICE);
  auto* layer2
      = dbTechLayer::create(tech2, "bottom_layer", dbTechLayerType::ROUTING);
  dbLib* lib1 = dbLib::create(db_database, "top_lib", tech1);
  dbLib* lib2 = dbLib::create(db_database, "bottom_lib", tech2);
  dbChip* chip = dbChip::create(db_database);
  dbBlock* block1 = dbBlock::create(chip, "top", tech1);
  dbBlock* block2 = dbBlock::create(block1, "bottom", tech2);
  // die size setting
  Rect die_rect(0, 0, 1000, 1000);
  block1->setDieArea(die_rect);
  block2->setDieArea(die_rect);

  vector<dbInst*> inst_collector_top;
  vector<dbInst*> inst_collector_bottom;

  // Example Library cell construction - top
  int top_lib_cell_num = 3;
  for (int i = 0; i < top_lib_cell_num; ++i) {
    int width = 100 + i * 10;
    int height = 200;
    dbMaster* master
        = dbMaster::create(lib1, ("top_cell_" + to_string(i)).c_str());
    master->setWidth(width);
    master->setHeight(height);
    master->setType(dbMasterType::CORE);

    int pin_num = 2 + i;
    for (int j = 0; j < pin_num; ++j) {
      dbSigType sig_type = dbSigType::SIGNAL;
      dbIoType io_type;
      if (j < pin_num - 1)
        io_type = dbIoType::INPUT;
      else
        io_type = dbIoType::OUTPUT;
      string pin_name = "top_pin_" + to_string(j);
      dbMTerm* master_terminal
          = dbMTerm::create(master, pin_name.c_str(), io_type, sig_type);
      dbMPin* master_pin = dbMPin::create(master_terminal);
      dbBox::create(master_pin, layer1, j * 2, j * 2, j * 2 + 1, j * 2 + 1);
    }
    master->setFrozen();
  }

  // Example Instance cells for top
  int instance_num = 10;
[... 266 lines cut ...]
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.30Bb4MFTdF
```

### To Reproduce

You can reproduce this error with this [commit](https://github.com/ApeachM/OpenROAD/commit/79a0c542fd4d1facbd65ea0bb717a71d54196d0c).

rovinski, 2023-08-27:
Resolved?

ApeachM, 2023-09-23:
Currently it is resolving now... I will let you know it is updated.

closed 2023-09-23 by ApeachM