The-OpenROAD-Project/OpenROAD #3789: OpenRoad crashes due to an assert

titan73, 2023-08-04:
### Describe the bug

I installed binaries from:
- https://github.com/Precision-Innovations/OpenROAD/releases/download/2023-07-31/openroad_2.0_amd64-ubuntu22.04-2023-07-31.deb
- https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-08-04/oss-cad-suite-linux-x64-20230804.tgz
- https://www.klayout.org/downloads/Ubuntu-22/klayout_0.28.10-1_amd64.deb

And got OFSR as of 392edae54a52586959855228f9fbfee50ebb56e4

When testing installation with make, openroad crashes due to an assert:

openroad: ./src/odb/src/db/dbWireCodec.cpp:481: void odb::dbWireEncoder::newPathVirtualWire(int, odb::dbTechLayer*, odb::dbWireType): Assertion `(jct_id >= 0) && (jct_id < _idx)' failed.
Signal 6 received
Stack trace:
0# 0x000055732DC089CD in /usr/bin/openroad
1# 0x00007F02D7242520 in /lib/x86_64-linux-gnu/libc.so.6
2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
3# raise in /lib/x86_64-linux-gnu/libc.so.6
4# abort in /lib/x86_64-linux-gnu/libc.so.6
5# 0x00007F02D722871B in /lib/x86_64-linux-gnu/libc.so.6
6# 0x00007F02D7239E96 in /lib/x86_64-linux-gnu/libc.so.6
7# odb::dbWireEncoder::newPathVirtualWire(int, odb::dbTechLayer*, odb::dbWireType) in /usr/bin/openroad
8# grt::RepairAntennas::addWireTerms(grt::Net*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, int, int, int, odb::dbTechLayer*, int, std::map<grt::RoutePt, std::vector<grt::Pin*, std::allocator<grt::Pin*> >, std::less<grt::RoutePt>, std::allocator<std::pair<grt::RoutePt const, std::vector<grt::Pin*, std::allocator<grt::Pin*> > > > >&, odb::dbWireEncoder&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in /usr/bin/openroad
9# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in /usr/bin/openroad
10# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in /usr/bin/openroad
11# ant::AntennaChecker::checkAntennas(odb::dbNet*, bool) in /usr/bin/openroad
12# ant::check_antennas(char const*, bool) in /usr/bin/openroad
13# 0x000055732E616A04 in /usr/bin/openroad
14# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
15# 0x00007F02DB808028 in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
18# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /usr/bin/openroad
19# ord::tclAppInit(Tcl_Interp*) in /usr/bin/openroad
20# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
21# main in /usr/bin/openroad
22# 0x00007F02D7229D90 in /lib/x86_64-linux-gnu/libc.so.6
23# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
24# _start in /usr/bin/openroad
Command terminated by signal 6
Elapsed time: 0:08.66[h:]min:sec. CPU time: user 0.75 sys 0.17 (10%). Peak memory: 190908KB.
make[1]: *** [Makefile:655: do-5_1_grt] Error 134
make[1]: Leaving directory '/media/sf_virtual_box/OpenROAD-flow-scripts/flow'
make: *** [Makefile:655: results/nangate45/gcd/base/5_1_grt.odb] Error 2

### Expected Behavior

This is a installation test as mentionned at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithPrebuilt.md

### Environment

```shell
It is a prebuilt installation as mentioned above.

kernel: Linux 5.19.0-32-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
```

### To Reproduce

make

vijayank88, 2023-08-04:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1306#issuecomment-1665349789 similar to this.
@vvbandeira is OpenROAD Debian package issue? 
Because locally built binaries working fine.

maliberty, 2023-08-04:
I suggest trying the latest build as there was a similar bug fixed recently.

titan73, 2023-08-08:
Crash is gone with latest .deb. Thanks!

titan73, 2023-08-08:
Also bug was similar to https://github.com/The-OpenROAD-Project/OpenROAD/pull/3757.