The-OpenROAD-Project/OpenROAD #3803: segfault in `check_antenna`

gadfort, 2023-08-06:
### Describe the bug

When running `check_antenna` OpenROAD segfaults.

```
 0# 0x000055C036B030AB in openroad
 1# 0x00007F98FDBB2520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbWireEncoder::addITerm(odb::dbITerm*) in openroad
 3# odb::tmg_conn::addToWire(int, int, int, bool, bool) in openroad
 4# odb::tmg_conn::treeReorder(bool) in openroad
 5# odb::tmg_conn::analyzeNet(odb::dbNet*) in openroad
 6# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in openroad
 7# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in openroad
 8# ant::AntennaChecker::checkAntennas(odb::dbNet*, bool) in openroad
 9# ant::check_antennas(char const*, bool) in openroad
10# 0x000055C0376BABD4 in openroad
```

### Expected Behavior

No segfault

### Environment

```shell
kernel: Linux 5.15.0-1040-azure
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.26.4
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9702-g92f8e07b0
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 118 lines cut ...]
Stack trace:

 0# 0x000055C036B030AB in openroad
 1# 0x00007F98FDBB2520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbWireEncoder::addITerm(odb::dbITerm*) in openroad
 3# odb::tmg_conn::addToWire(int, int, int, bool, bool) in openroad
 4# odb::tmg_conn::treeReorder(bool) in openroad
 5# odb::tmg_conn::analyzeNet(odb::dbNet*) in openroad
 6# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in openroad
 7# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in openroad
 8# ant::AntennaChecker::checkAntennas(odb::dbNet*, bool) in openroad
 9# ant::check_antennas(char const*, bool) in openroad
10# 0x000055C0376BABD4 in openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F9901A4A028 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
16# ord::tclAppInit(Tcl_Interp*) in openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in openroad
19# 0x00007F98FDB99D90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in openroad
```

maliberty, 2023-08-07:
I find these SiC packaged test cases hard to follow.  How do I determine exactly what files/commands are being executed?

gadfort, 2023-08-07:
@maliberty 
The openroad scripts are located in: build/ibex/job0_freepdk45_demo/sc_collected_files/scripts_fccbfa790db5ddcc0abfcccac3b21bc686d7535a/

All the files needed to run this are placed in build/ibex/job0_freepdk45_demo/sc_collected_files

The run.sh script is simply going to call:
```
./build/ibex/job0_freepdk45_demo/route/0/replay.sh
```
which is the script that executes OpenROAD

The two files are going to be: `sc_apr.tcl` and `sc_route.tcl`

maliberty, 2023-08-16:
@eder-matheus what is your status?

eder-matheus, 2023-08-17:
@maliberty I'm trying to isolate the problem, to understand on which net the crash happens and how the wires for it are, but I'm having trouble reproducing the crash. Sometimes it just goes through the routing flow without errors, but I'm also having this error:
[image]

I'm running in a MacOS, with the latest openroad compiled in debug mode. I'll try it on CentOS too.

maliberty, 2023-08-17:
Please try with valgrind

gadfort, 2023-08-17:
@eder-matheus the stuck issue seems to be from the boost stracktrace getting into a deadlock with a atomic lock

gadfort, 2023-08-20:
@eder-matheus I've narrowed it down a little.
If I exclude the net `rst_ni` from checking and from wire creation, doesn't segfault. It looks like this net has a fanout of 1700 (it's a port), so the wire that GRT is encoding is probably corrupted (so there probably is a bug in the GRT wire encoding code).

This seems to be from not having a driving cell set on the inputs or a max_fanout causing the `repair_design` to ignore the net. @maliberty I can fix the constraints since they are wrong, but to seems like there should be some kind of warning either in `check_setup` or `repair_design`? Not sure what the trigger should be, but no cell in the library can drive this kind of fanout.

eder-matheus, 2023-08-21:
> @eder-matheus I've narrowed it down a little. If I exclude the net `rst_ni` from checking and from wire creation, doesn't segfault. It looks like this net has a fanout of 1700 (it's a port), so the [...]

Thanks for the update, @gadfort. The main issue is how we evaluate the global routing to search for antenna violations. The GRT tool does not create local connections with the pins, just the connection between the guides. When we evaluate the antennas, we convert the guides to wires and create the connection with the pins. In these connections, we can have overlaps between the wires and pin shapes, leading to the crash in order_wires.

We already made some improvements on this conversion process, trying to avoid the overlaps, but it may be hard in a scenario like yours, with a high fanout net.

I'm wondering if we should change this approach and, instead of converting the groute to actual wires, make the antenna checker understand the route format and evaluate it.

For now, I'll dig at this net you mentioned to understand precisely what issue order_wires is having trouble with.

gadfort, 2023-08-21:
@maliberty thats a good point, I thought it was turned on in that run, but certainly didn't seem to occur. But I would still advocate for some warnings if that is possible.

gadfort, 2023-08-21:
yeah, `check_setup` would be ideal, 
Warn on:
1. set_driving_cell or set_max_fanout on input_ports
2. set_load on output_ports
I'm not sure what the right combinations would be, but I wouldn't generate a warning if driving cell is set and max_fanout is not.

maliberty, 2023-08-21:
Please open an enhancement request for this. 

For (1) I wouldn't warn if you have max cap/slew set as those effectively limit the fanout as well.

gadfort, 2023-08-21:
Done https://github.com/The-OpenROAD-Project/OpenROAD/issues/3884

maliberty, 2023-10-17:
@eder-matheus was this crash fixed?

closed 2023-12-05 by eder-matheus