The-OpenROAD-Project/OpenROAD #3806: Unknown layer for viarule Error in the detailed route stage.

faisaladilquadery, 2023-08-07:
### Describe the bug

while trying to integrate the design 'or1200_immu_top' with a manual integrated pdk 'gpdk45', I have been facing this error 'Unknown layer Oxide for viarule M1_DIFF' while sourcing the detail_route.tcl in the route stage. Can anyone suggest any possible solution?

[image]
Lef syntax example : 

[image]

[image]

The layer poly is defined in the same way as oxide but doesnt throw an error : 

[image]

[image]

### Expected Behavior

This design is running in other pdks but not running in this one.

### Environment

```shell
.
```

### To Reproduce

.

vijayank88, 2023-08-07:
@faisaladilquadery 
The pdk is public or private?

faisaladilquadery, 2023-08-07:
> @faisaladilquadery The pdk is public or private?

The pdk is private.

maliberty, 2023-08-07:
Do you have more than one MASTERSLICE layer?

maliberty, 2023-08-07:
As a short term workaround you can remove that viarule as we won't generate such vias anyways.

osamahammad21, 2023-08-07:
@maliberty I think this is the tip of the iceberg. Because we log an error if we find any object on a layer that is not considered in drt database. So that could happen for any other object, obstruction or pin shapes for example. What should we do in that case?

maliberty, 2023-08-07:
It looks like the layer is in LEF so I'm assuming it is in odb but not passing through drt/io.

osamahammad21, 2023-08-07:
It is in odb, but we don't consider all layers in odb. We consider the first masterslice layer only.

faisaladilquadery, 2023-08-08:
> As a short term workaround you can remove that viarule as we won't generate such vias anyways.
I already tried that, the stage passes but then gives this error while reading the design :
[image]