The-OpenROAD-Project/OpenROAD #3817: A lot of slew in asap7/sram-64x16

maliberty, 2023-08-08:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3811

<div type='discussions-op-text'>

<sup>Originally posted by **oharboe** August  7, 2023</sup>
Is this amount of slew expected on the clock pin to the SRAM? 

`SRAM2RW16x32_2/CE2` is one of two clocks to the SRAM, the SRAM has one read and one write port.

```
make DESIGN_CONFIG=designs/asap7/sram-64x16/config.mk
```

```
>>> report_checks -path_delay max -fields {slew net cap} -to _351_/D
Startpoint: SRAM2RW16x32_2 (rising edge-triggered flip-flop clocked by io_clk)
Endpoint: _351_ (falling edge-triggered flip-flop clocked by io_clk')
Path Group: io_clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock io_clk (rise edge)
                        241.44  241.44   clock network delay (propagated)
                150.83    0.00  241.44 ^ SRAM2RW16x32_2/CE2 (SRAM2RW16x32)
                 40.24  202.07  443.51 v SRAM2RW16x32_2/O2[4] (SRAM2RW16x32)
     1    9.73                           _SRAM2RW16x32_2_O2[4] (net)
                 53.97    5.14  448.65 v _351_/D (DFFLQNx1_ASAP7_75t_R)
                                448.65   data arrival time

                        500.00  500.00   clock io_clk' (fall edge)
                        186.47  686.47   clock network delay (propagated)
                          0.52  686.99   clock reconvergence pessimism
                                686.99 v _351_/CLK (DFFLQNx1_ASAP7_75t_R)
                        -22.88  664.11   library setup time
                                664.11   data required time
-----------------------------------------------------------------------------
                                664.11   data required time
                               -448.65   data arrival time
-----------------------------------------------------------------------------
                                215.46   slack (MET)
```
</div>

maliberty, 2023-08-08:
```report_checks -path_delay max -fields {slew net cap} -to _351_/D```

maliberty, 2023-08-08:
@mynnguyen you can run it yourself and add the needed options to get more info

tspyrou, 2023-09-28:
@mynnguyen opensta support the full clock expanded report which will show you the clock tree and the driving cell of the sram.

precisionmoon, 2023-09-28:
Correct.  It seems slew degradation is due to long clock wires.

[image]

A new option clock_tree_synthesis -obstruction_aware was checked in yesterday that considers impact of blockages and hard macros.  This option reduces slew and delay of clock buffers, but not all buffers see improvement in both delay and slew.

[image]

Final setup TNS is reduced by 49% from -82.55 to -41.88 with -obstruction_aware option.
Legalizer displacement after CTS shows improvement with this option also.

[image]

Is this obstruction aware option partially helpful?

oharboe, 2023-09-29:
@precisionmoon @tspyrou Why is it an option? It would be better if the tool always did the best thing without the user having to be aware of the option and have a mental model of when it should be applied.

QuantamHD, 2023-09-29:
+1 @oharboe should be default, or auto disabled if there's no benefit

oharboe, 2023-09-29:
I know that any new behavior will make *some* design very slightly worse and that the test baseline has to be updated.  I hope this work is not creating a bias towards not having the right defaults and behaviors of OpenROAD. :sweat_smile:

precisionmoon, 2023-09-29:
Yes, we absolutely agree to make this on by default.   We are still tuning the algorithm, so we want to avoid QoR fluctuations in the short term.   We expect to be done with tuning in a few weeks.   After that, the feature will be on by default.  In the mean time, we welcome your feedback on the feature.

Thank you for your patience.

rovinski, 2023-09-29:
I think we could make this process more transparent by flagging features as "experimental" in docs until they are ready for use.

closed 2025-11-17 by precisionmoon