The-OpenROAD-Project/OpenROAD #3820: Running cmake issue an error about or-tools

titan73, 2023-08-08:
### Describe the bug

When building OpenRoad with "cmake .." from build directory. I get an error about missing or-tools.

This is a bit odd the first I've been able to do:
- cmake ..
- make
And the problem started appearing when doing "make install" and now it appears every time.

 

### Expected Behavior

Finish configuration.

### Environment

```shell
See output as Env.sh gives something similar to cmake.
```

### To Reproduce

cd build
cmake ..

### Relevant log output

```shell
-- TCL header: /usr/include/tcl/tcl.h
CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  By not providing "Findortools.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ortools", but
  CMake did not find one.

  Could not find a package configuration file provided by "ortools" with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
  "ortools_DIR" to a directory containing one of the above files.  If
  "ortools" provides a separate development package or SDK, be sure it has
  been installed.
```

titan73, 2023-08-08:
Ok I figured out what happened. I deinstalled the openroad .deb before doing "make install".
So basically the problem occurs when the .deb is not installed so I guess or-tools stuff are installed with openroad binaries.

vijayank88, 2023-08-09:
@titan73 
Can you point link which you referring for OpenROAD installation?

titan73, 2023-08-09:
> @titan73 Can you point link which you referring for OpenROAD installation?

The installation guide is at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithPrebuilt.md

The prebuilt openroad package is at https://github.com/Precision-Innovations/OpenROAD/releases.

The related package is at https://github.com/Precision-Innovations/OpenROAD/releases/download/2023-07-26/openroad_2.0_amd64-ubuntu22.04-2023-07-26.deb

vvbandeira, 2023-08-09:
@titan73 
If you uninstalled the `.deb` and want to compile from the source you will need to follow these instructions:
https://openroad.readthedocs.io/en/latest/user/Build.html#install-dependencies

titan73, 2023-08-09:
This is the very link as followed to compile. So basically I did:
- git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
- cd OpenROAD
- sudo ./etc/DependencyInstaller.sh
- mkdir build && cd build
- cmake ..
I guess or-tools installation should be added to etc/DependencyInstaller.sh

vvbandeira, 2023-08-09:
It is included, see: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh#L150

Could you share the output of `etc/Env.sh`? This gives additional information that is useful.

titan73, 2023-08-10:
Ah indeed it's there. I did a bad grep I guess.
Here is the output of etc/Env.sh
[env.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/12312350/env.log)

titan73, 2023-08-10:
Finally I figured out what happened. When I removed openroad .deb, or-tools was removed too as it's bundled with openroad.
Running again etc/DependencyInstaller.sh reinstalled or-tools and the problem is gone.
Thanks!

closed 2023-08-10 by titan73